Ultimaker/Cura #1960: Open Projects - creates New Duplicate Printers

changedsoul, 2017-06-10:
If you open a saved Project with models, cura will create a new Printer. If you open the exact same project again, cura will create yet another printer, naming them Printer #1, #2, #3, etc. Soon you have a whole slew of added printer you need to go back and delete :)

Ghostkeeper, 2017-06-11:
It should be detecting that you already had a printer with the specified definition, and re-use that one.

I'm not creating a bug report out of this one yet though because we changed a few things after the release of the 2.6 beta with this code, and it might've been fixed since then. I can't test with a more recent build at the moment.

changedsoul, 2017-06-11:
Is there a newer build of the beta I can test? Mine just says 2.6.0-BETA. I am truly digging this OctoPrint implementation in Cura. I just set up octoprint right before I tried the Beta, not knowing it would be there. Much to my surprise I could print directly to my printer using OctoPrint....So Awesome! I dont want to go back to 2.5, lol

fieldOfView, 2017-06-11:
>  I am truly digging this OctoPrint implementation in Cura.

Off-topic, but that is nice to hear

changedsoul, 2017-06-12:
I think I understand what you mean about detecting the printer. I opened a project created under 2.6 Beta and it did detect the printer fine, and did not add another one. 
Opening a project from 2.5 though did add a new one.

uwe999, 2017-06-12:
I had the same issue with 2.6 beta when opening a project. It created another printer with the same same and "#2" appended to the name

nallath, 2017-06-12:
I just tried this; If you open a project it shows you a window. There should be a dropdown menu there next to the machine. It should say either "Update Existing" or "Create new" 

Depending on that choice it either creates a new one or updates the existing machine.

changedsoul, 2017-06-13:
These drop downs to exist when opening a project created in 2.6 Beta, but not when opening a previous versions Projects

LipuFei, 2017-06-14:
I have tested using 2.5 projects and 2.6-beta projects with the current 2.6 branch, and the problem seems to be gone. Yesterday we made a fix (https://github.com/Ultimaker/Uranium/commit/c91230e539a9ae0e2cf4cd9e96909ef425becb76) for something else, which fortunately also fixed this issue. :)

closed 2017-06-23 by Ghostkeeper