Ultimaker/Cura #1962: No brim when enabling supports

bipsendk, 2017-06-11:
In Cura 2.6.0 Beta   - when slicing the front brace STL file from https://www.thingiverse.com/thing:2118713 and choosing to have a brim around the printed object, brims are only put where the supports are constructed.

[image]

Ghostkeeper, 2017-06-19:
Doesn't happen here: 
[image]

Ghostkeeper, 2017-06-19:
Perhaps the model wasn't entirely aligned to the build plate? If it's floating in the air, only the support will be touching the first layer.

BagelOrb, 2017-07-14:
Caused by the same issue as https://github.com/Ultimaker/Cura/issues/2050

closed 2017-07-14 by BagelOrb