Ultimaker/Cura #1966: "open project" dialog box too small

uwe999, 2017-06-12:
cura 2.6.0 beta:
when opening a project I get this dialog box with the text cut off:
[image]

nallath, 2017-06-14:
Ugh, not this again :( What is your operating system, resolution & screen?

uwe999, 2017-06-14:
win10, 1920x1080 and "change the size of text, apps, and other items: 150%
(recommended)"

Ghostkeeper, 2017-09-18:
Our project manager just kicked this off our planning because the ticket was older than 12 weeks.

fieldOfView, 2017-09-18:
Sure, you can ignore it. Doesn't mean it is not broken
https://github.com/Ultimaker/Cura/commit/7bc715a9bd5e61efcfef4b54f7d004b3f6248f5a

I have said this before, and several others have too. Specifying sizes in pixelsizes without taking Screen.devicePixelRatio is not correct on Windows when the user sets the "Change the size of text, apps and other items" to anything but 100%. On high-ish DPI screens the default is often 150% or even 200%. It worked correctly before, but then the Screen.devicePixelRatio was removed willy-nilly in a number of QML files, breaking those windows.

fieldOfView, 2017-09-27:
Should be fixed now, can be closed.
https://github.com/Ultimaker/Cura/pull/2484

closed 2017-09-28 by nallath