Ultimaker/Cura #1967: DisplayLink drivers not detected, Failed to probe OpenGL.

briandiephuis, 2017-06-12:
Application Version: 2.x
Platform: Windows 10 Home 64-bit
Qt: ?
PyQt: ?
Display Driver: DisplayLink USB Device (http://www.displaylink.com)

Steps to Reproduce:
Connect a Dell D3100 docking station and use only the display(s) on the docking station (i.e. disable the laptop display). Open Cura 2.x (tested with Cura 2.4, 2.5 and 2.6) and it will not start.

Actual Results:
Cura does not start, instead this message shows: "**Failed to probe OpenGL** Could not probe OpenGL. This program requires OpenGL 2.0 or higher. Please check your video card drivers."

Expected results:
Cura should start like it normally does. The message only shows only when _just_ the DisplayLink display(s) are used. Enabling the laptop display is enough to tell Cura that all is well and OpenGL is actually working on this machine. Cura will just keep on working when the laptop display screen is disabled after starting it.

Additional Information:
I use a dual monitor setup connected to a USB docking station on my Dell XPS 13 (9343) and I normally don't use the laptop display. I think the DisplayLink drivers are not (well) recognised. If this is a problem at their side, please let me know.

This is not a high priority bug, as I can just enable the laptop display temporarily, but the little ultrabook has a hard time driving three displays, so it would be nice if this was not needed.

nallath, 2017-06-12:
I think this is a driver issue on their end.

awhiemstra, 2017-06-12:
We basically just create an OpenGL context for the current window. If that fails for some reason, you get the resulting error. It seems either Qt (which we use for context creation) or a lower level is doing something weird here.

briandiephuis, 2017-06-14:
O well, you can close the issue then. It's probably a very rare use-case any way and if there is no (easy) fix from your side then I'll just enable the laptop screen temporarily when I start Cura :)

closed 2017-06-23 by Ghostkeeper

ChrisTerBeke, 2018-01-08:
Maybe this helps: http://support.displaylink.com/knowledgebase/articles/543922-games-do-not-work-on-windows-with-displaylink-soft ?