Ultimaker/Cura #1968: Model not flat on build plate

uwe999, 2017-06-12:
Cura 2.6.0 beta:
I had several failed print on this T-rex skull enlarged to 120%. I have my bed leveled and print on PEI and rarely have an issue with a print coming unstuck during printing.
After doing some more debugging it turns out that the model isn't flat on the build plate after enlarging to 120 percent. To duplicate:
Drag & drop https://www.thingiverse.com/download:550809 to the build plate. Enable brim and supports. The "layer view" shows that supports and first layer are printed with brim.
Now enlarge to 120%. "Layer view" now shows that main model isn't printed on first layer.

I wasn't able to duplicate this issue with a more simple stl.

[image]

smartavionics, 2017-06-12:
Hi, it worked OK for me using the current master branch...
[image]

ChrisTerBeke, 2018-03-02:
please re-open if it's really still an issue (but I can't reproduce it anywhere).

closed 2018-03-02 by ChrisTerBeke

Ghostkeeper, 2018-03-28:
This was fixed in release Cura 3.2 (or at least a bug that looks very much like it).