Ultimaker/Cura #1969: Tab Through Fields

CCS86, 2017-06-12:
Hi guys, would it be a pain to allow tabbing through the setting fields? 

It would really be nice to tab through when setting up a print!

fieldOfView, 2017-06-12:
I agree that would be nice. I remember a couple of small complications from when I last tried to implement this. For example all the controls will need a "has keyboard focus" state. Not all controls respond to keyboard input at the moment (eg checkboxes, section headers).

DDDirk, 2017-06-13:
Keyboard control is more than just nice. At the moment my RSI is really bad, to the extent that it really hinders me in my work (reduced control over finger movements). Tabbing through the fields is just one thing. Simply using the menu in a standard way is something that would also help. (So all menus accessible with Alt-<key> and moving through the menu structure with the arrow keys.)

nallath, 2017-07-27:
This has been implemented.

closed 2017-07-27 by nallath

CCS86, 2017-07-29:
@nallath , which version? 

I just installed 2.6.2, and don't have that functionality.

fieldOfView, 2017-07-30:
Cura 2.7, which has not yet been released. I expect a beta to be available in a week or so

CCS86, 2017-12-19:
Is it possible to change the behavior so that the current field value is selected when tabbing I to the field?

This would make it so much more efficient to tab through, quickly keying values, instead of having to delete every current entry value first.

fieldOfView, 2017-12-19:
I have made a PR (#3017) to implement this. It is something you can try/test for yourself by editing resources/qml/Settings/SettingTextField.qml (it is a single line addition)

CCS86, 2017-12-27:
Awesome, thanks @fieldOfView , that worked perfectly!