Ultimaker/Cura #1970: Cura not rendering

CamoJackson, 2017-06-12:
On 2.5.0 Cura isn't rendering in the window. Windows Build 16215.1000 version 1703.
[image]

fieldOfView, 2017-06-12:
We need more information. Please read the section "Logging Issues" here: 
https://github.com/Ultimaker/Cura/blob/master/README.md

CamoJackson, 2017-06-12:
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/1069426/DxDiag.txt)
and I'm not getting a log for 2.5 or 2.6. both are not rendering.

nallath, 2017-06-13:
Could you also share the Cura logs?

Ghostkeeper, 2017-06-26:
Also worth trying: Update video drivers.

alekseisasin, 2018-05-16:
@CamoJackson , Do you have any similar problem with the latest Cura version ?

ianpaschal, 2018-05-22:
I'm taking that as a no.