Ultimaker/Cura #1971: Preheat non-functional

michaelpond08, 2017-06-12:
Ultimaker 3 Extended, connected via wifi. Cura 2.5.0 Clicking on the "Pre-heat" button on the printer monitor screen causes the build plate setpoint to change briefly to desired temperature, then reverts to 0 deg C.

nallath, 2017-06-14:
Could you share the logs?

michaelpond08, 2017-06-19:
Ok, this may be a stupid question, but I looked on the "C" drive on my computer for about 20 minutes, and could not find the log files. Would someone be able to direct me to those files so I can share them?

nallath, 2017-06-20:
help-> show configuration folder. There should be a cura.log file there.

michaelpond08, 2017-06-20:
hmmm. couldn't load the file. This is the line it puts in the file every time I click the "Pre-heat" button

2017-06-20 07:54:47,639 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevice._onFinished [1090]: Something went wrong when trying to update data of API (http://192.168.1.103/api/v1/printer/bed/temperature/target). Message: b'{"message": "Method not allowed"}' Statuscode: 405

Ghostkeeper, 2017-06-26:
I've noticed before that on more recent versions of the UM3 firmware, the pre-heat functionality doesn't seem to work in all of the cases where it previously did. It does work when idling (for me) but for instance it doesn't work any more when the printer is waiting for the build plate to clear.

Ghostkeeper, 2017-06-30:
Perhaps your firmware is outdated, by the way?

michaelpond08, 2017-06-30:
Just checked that, and it was 2 versions back. Updated the firmware, and it works like a charm!

Thank you for your help!

closed 2017-06-30 by michaelpond08