Ultimaker/Cura #1972: Profiles and Material confusing

ogoudge, 2017-06-13:
The linking of profiles and materials is confusing and appears to be broken, or possible needs to be more clear and simple.

1. It be nice to be able to remove the default material and build my own with my own retraction settings etc.
2. When making the changes in the material also found I need to build a new profile. However this still did not work.
3. To have both is to redundant.

The issue I have a print that I need 100% fill on, The best I can get is about 10% even though all my setting show 100% someplace along the line its pulling settings from someplace you can not see.

To me having this flaw or poor flow makes this version of the software not even usable.

One simple fix for this would be to set any changes made in the custom setting override any material or profile settings. Personally I think you should just drop those features all together or give an option to not use them and allow people to simple set there perimeters under the costume settings.

fieldOfView, 2017-06-13:
You can change the "has_materials" metadata of your printer definition to false. This will remove material selection for that printer.

> One simple fix for this would be to set any changes made in the custom setting override any material or profile settings.

That should already be the case.

nallath, 2017-06-14:
Could you share a project file of your setup? It seems like something is going wrong.

ogoudge, 2017-06-16:
I had to install and resetup the software here is a project.

ogoudge, 2017-06-17:
I do not see this "has_materials" I have also tried to set the defaults but they are still settings from someplace. It really seams unnecessary to have Material, profile, and print setup. Should lose profile keep material and make it so print setup options override profile options.

fieldOfView, 2017-06-17:
All definitions inherit from fdmprinter.def.json. This definition sets the "has_materials" metadata to true, and there are very few definitions that don't use materials, so you do not often see a "has_materials" metadata entry. But you can add it to the "metadata" section of the definition file.

See the Moai definition for an example:
https://github.com/Ultimaker/Cura/blob/master/resources/definitions/peopoly_moai.def.json#L13

NB: pay close attention to the syntax of the json file, specifically the comma at the end of the line. The last line in the metadata section should not have a comma, the others do.

Ghostkeeper, 2017-06-29:
> I had to install and resetup the software here is a project.

The project seems to have gone lost. If you post that as an attachment via the e-mail it won't get attached to the issue here. That attachment gets lost.

ianpaschal, 2018-02-27:
OP never responded about the missing project, so I am closing this.

closed 2018-02-27 by ianpaschal

ogoudge, 2018-02-27:
Yea quit using cura that was the fix

Sent from BlueMail<http://www.bluemail.me/r?b=12095>