Ultimaker/Cura #1973: Camera wont flip views

abetterway, 2017-06-13:
I am unable to find a way to watch my Raspberry Pi camera without physically flipping it over or moving it.
I have several text objects in the cameras field of view making this particularly troubling for me.
Fortunately Octoprint has check boxes to remedy this using their GUI in a browser.

fieldOfView, 2017-06-14:
The OctoPrint plugin is a separate project: https://github.com/fieldOfView/OctoPrintPlugin

As luck would have it, I have just implemented this feature here:
https://github.com/fieldOfView/OctoPrintPlugin/issues/29
You will have to wait for Cura 2.7 because it relies on some recent changes in Cura that are not part of the Cura 2.6 release.

abetterway, 2017-06-17:
Thanks for being on the ball!

closed 2017-06-17 by nallath

fieldOfView, 2017-08-02:
A beta of Cura 2.7 is now available. I would appreciate if you could test this functionality