Ultimaker/Cura #1977: Machine settings build surface with multi extruders

maukcc, 2017-06-15:
Cura 2.5.99 master
OS Ubuntu 16.04

When not printing with T0:
Setting a nozzle offset, makes a disallowed area on both sides of the build surface and not 1.
When you set a negative offset, it makes a positive disallowed area on the build surface.

Switching from 2 nozzles to 1, remembers the offset and gives you a disallowed area

Let the machine settings dialog box make it clear that the X,Y,Z length settings are for maximum travel of the axis and not the bed size.

Ghostkeeper, 2017-06-28:
Fixed in fb1d7bc22363d3a3550748c1180d43ceba444ff0.

maukcc, 2017-06-28:
Fantastic ! that did the job. I have not tried printing, and will give it a go in a sec.

1 thing that is not correct, is that it still incorporates the disallowed area of extruders that are not there.
Our machine defenition has 4 extruders and standard only 2 are visible, but it respects the disallowed area of the other 2 as well.

Ghostkeeper, 2017-06-28:
Oh, are you sure they aren't used? Not for the platform adhesion or something? Or infill extruder?

We detect the used extruders here: https://github.com/Ultimaker/Cura/blob/master/cura/Settings/ExtruderManager.py#L416

maukcc, 2017-06-28:
I can not set them in Cura, as it will crash instantly. See: #1948

so we added it to the definition file:
        "wall_extruder_nr": { "default_value": 0 },
        "wall_0_extruder_nr": { "default_value": 0 },
        "wall_x_extruder_nr": { "default_value": 0 },
        "top_bottom_extruder_nr": { "default_value": 0 },
        "infill_extruder_nr": { "default_value": 0 },

But it still gives me disallowed areas generated by unused extruders.
With unused I do not mean that they are not used for this print job, but they are not selected in "machine settings". So extruder_count is at 2 and not 4.

maukcc, 2017-06-30:
Also setting the disallowed areas in printer.def.json goes in the wrong direction in Y axis.
If you give it a negative value, you would expect it to come to the front.

Ghostkeeper, 2017-07-03:
Oops, it seems I forgot one. Fixed: https://github.com/Ultimaker/Cura/commit/319559740dd8c8c2f9ffad637bee43cef9345073

Ghostkeeper, 2017-07-03:
> With unused I do not mean that they are not used for this print job, but they are not selected in "machine settings". So extruder_count is at 2 and not 4.

I'd think that whatever extruders are unused per the extruder count should also be unused in the actual print...

maukcc, 2017-07-03:
I agree, but it is not being done :)

So at this time all disallowed areas from all extruders are being generated. 
- does not matter if they are used in the printjob
- does not even matter if they are used in machine settings.

And the disallowed areas in printer.def.json go in the wrong direction in Y axis.

3195597 was also done in  fb1d7bc.

Ghostkeeper, 2017-07-07:
Yes, 319559740dd8c8c2f9ffad637bee43cef9345073 was reverted.

Ghostkeeper, 2017-07-07:
What should still be done for this?

> And the disallowed areas in printer.def.json go in the wrong direction in Y axis.

...seems to contradict...

> 3195597 was also done in fb1d7bc.

nallath, 2017-07-07:
It was reverted because the inversion already happened at a few layers lower and it re-introduced a crash for single extrusion machines.

Ghostkeeper, 2017-07-10:
Indeed, I know that, but

> the inversion already happened at a few lines lower

...seems to contradict...

> And the disallowed areas in printer.def.json go in the wrong direction in Y axis.

It was already inverted, but the disallowed areas still go in the wrong direction? So it shouldn't be inverted?

fieldOfView, 2017-07-10:
> It was already inverted, but the disallowed areas still go in the wrong direction?

Then perhaps try... not inverting it? (in line 734, where the values has already been checked for not being None)

maukcc, 2017-07-12:
Also the move tool in the GUI moves the Y in the wrong direction

maukcc, 2018-02-09:
We are at V3.2.0 and it still persists. The most obvious one is in the GUI: If you move your object 20mm in Y, it goes -20.
I can not imagine this to be a very hard to solve bug?

Ghostkeeper, 2018-03-05:
No, it's not, but nobody has taken a look at it since.

Back then I was confused by people saying that the Y-direction was inverted by my commit and that was wrong, and before I inverted it it was also wrong. My commit was reverted because I had introduced a crash for single extrusion machines. Now I see that the Y offset itself was correct but the direction in which it shifts when switching extruders was wrong. Let's hope this will correct that: 396ea274c457dc74005b40a2897d5d9c37539451

closed 2018-03-05 by Ghostkeeper

maukcc, 2018-03-13:
This still persists. To reproduce, place an object on the buildplate then tell it to move +20mm in Y direction(away from you). It will then move 20mm in the minus direction(towards you).

maukcc, 2018-04-04:
This still persists.

commit 396ea274 (2018-03-05): Invert Y direction of nozzle offsets

If there is a Y offset it should now move the disallowed areas in the correct direction

The initial placement of the disallowed borders was already correct.

Fixes #1977.