Ultimaker/Cura #1979: Parts do not snap to ground

mitchbarbon, 2017-06-15:
When parts are brought into Cura, they sometimes **_do not snap to ground plan_**. The attached image show this perfectly with three different part files. When the part is above the surface, Cura allows you to "eyeball it" and move the part down. If it is below the surface, **_you cannot move the part upward_**. The parts are scaled in the image, but this issue occurs before converting to inches. I have also tried drag and drop, versus opening the file through Cura. To my knowledge, this is happening in at least the previous few generations of Cura including its most recent release. We use Creo PTC for modeling in case that means anything.
[image]

nallath, 2017-06-15:
Can you share the part? I think this is something about this model that is causing issues.

mitchbarbon, 2017-06-16:
I can share this part.
[image]

[PRT0003.zip](https://github.com/Ultimaker/Cura/files/1080674/PRT0003.zip)

Ghostkeeper, 2017-06-28:
Seems that this happens when it auto-scales the model. Because this model was modelled in a different unit, Cura auto-scaled it to something that's more logical (it displays a message that it did so). If I rescale it in CloudCompare and then load the model in Cura it works correctly.

Ghostkeeper, 2017-10-02:
Our project manager just removed this bug from our planning because it's more than 12 weeks old.

ChrisTerBeke, 2017-11-14:
Is this still an issue?

mitchbarbon, 2017-11-14:
Hello,

This is not an issue if the part is in mm. But we always work in inches so this is still a problem if we do not convert the file to mm before exporting.

Thanks,
Mitch

Ghostkeeper, 2017-11-24:
We've been experimenting with a fix yesterday for these (there are a lot of duplicates of this issue). We found something but it might have a performance impact so we're testing that a bit.

Ghostkeeper, 2017-12-15:
This is fixed for Cura 3.2.

closed 2017-12-15 by Ghostkeeper

mitchbarbon, 2017-12-15:
Thanks! This will be such great help!