Ultimaker/Cura #1981: Z hop feedrate

maukcc, 2017-06-16:
The command to make the Z hop, does not have a feedrate specified. So it would be something like G1 Z1.3
So now it is depending on the feedrate set by a previous command. 
This could be ultra fast or ultra slow, which then takes forever to move it.

Option would be to give it the max_feedrate_z_override parameter, or a separate z_hop_feedrate parameter

BagelOrb, 2017-06-27:
But the maximum Z feedrate is already set by M203 Z[max_feedrate_z_override].
The firmware should already enforce the max Z speed.

If you want the Z hops to have a higher speed than the last XY move it does make sense to add a separate z_hop_speed setting, though.

BagelOrb, 2017-06-27:
Did you actually experience problems with this? Or did you just notice it was a problem by looking at the gcode?

maukcc, 2017-06-27:
Max speed is not a big issue, as there is a setting for that in Cura, and it is limited in firmware.

The thing we are having an issue with, is that we have a material that needs very slow retraction, and thus the following Z-hop is also in that very slow fashion, resulting in a Z-hop that takes 10 seconds.

BagelOrb, 2017-06-27:
Haha wow!

I'll see what I can do.

BagelOrb, 2017-06-27:
Fixed: https://github.com/Ultimaker/CuraEngine/commit/78d7c0630a9784860aadbc644b2e4d81c9eca12b

closed 2017-06-27 by BagelOrb

maukcc, 2017-06-28:
Fantastic ! that did the job. I have not tried printing, and will give it a go in a sec.

1 thing that is not correct, is that it still incorporates the disallowed area of extruders that are not there.
Our machine defenition has 4 extruders and standard only 2 are visible, but it respects the disallowed area of the other 2 as well.