Ultimaker/Cura #1982: CRITICAL - cura.CrashHandler.show [42]:     if self.errorCallback:

smartavionics, 2017-06-16:
2017-06-16 11:53:07,238 - CRITICAL - cura.CrashHandler.show [39]: An uncaught exception has occurred!
2017-06-16 11:53:07,238 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] 
2017-06-16 11:53:07,239 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] This program is free software: you can redistribute it and/or modify
2017-06-16 11:53:07,239 - CRITICAL - cura.CrashHandler.show [42]: Traceback (most recent call last):
2017-06-16 11:53:07,242 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] it under the terms of the GNU Affero General Public License as published by
2017-06-16 11:53:07,242 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/local/lib/python3/dist-packages/UM/Backend/SignalSocket.py", line 56, in error
2017-06-16 11:53:07,242 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] the Free Software Foundation, either version 3 of the License, or
2017-06-16 11:53:07,243 - CRITICAL - cura.CrashHandler.show [42]:     if self.errorCallback:
2017-06-16 11:53:07,243 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] (at your option) any later version.
2017-06-16 11:53:07,243 - CRITICAL - cura.CrashHandler.show [42]: AttributeError: '_SocketListener' object has no attribute 'errorCallback'
2017-06-16 11:53:07,244 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] 
Fatal Python error: Segmentation fault

Thread 0x00007f2f6de96700 (most recent call first):
  File "/usr/lib/python3.5/logging/__init__.py", line 984 in emit
  File "/usr/lib/python3.5/logging/__init__.py", line 855 in handle
  File "/usr/lib/python3.5/logging/__init__.py", line 1487 in callHandlers
  File "/usr/lib/python3.5/logging/__init__.py", line 1425 in handle
  File "/usr/lib/python3.5/logging/__init__.py", line 1415 in _log
  File "/usr/lib/python3.5/logging/__init__.py", line 1267 in debug
  File "/usr/local/lib/uranium/plugins/ConsoleLogger/ConsoleLogger.py", line 42 in log
  File "/usr/local/lib/python3/dist-packages/UM/Logger.py", line 60 in log
  File "/usr/local/lib/python3/dist-packages/UM/Backend/Backend.py", line 90 in _backendLog
  File "/usr/local/lib/python3/dist-packages/UM/Backend/Backend.py", line 153 in _storeStderrToLogThread
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f2f6c693700 (most recent call first):
  File "/usr/local/lib/python3/dist-packages/UM/Backend/Backend.py", line 142 in _storeOutputToLogThread
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f2fa8bd7700 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 293 in wait
  File "/usr/lib/python3.5/threading.py", line 424 in acquire
  File "/usr/local/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/local/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f2fc1f46700 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 293 in wait
  File "/usr/lib/python3.5/threading.py", line 424 in acquire
  File "/usr/local/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/local/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f2fcfcbb700 (most recent call first):
  File "/usr/local/lib/python3/dist-packages/UM/Job.py", line 109 in yieldThread
  File "/usr/local/lib/cura/plugins/CuraEngineBackend/StartSliceJob.py", line 252 in _buildGlobalSettingsMessage
  File "/usr/local/lib/cura/plugins/CuraEngineBackend/StartSliceJob.py", line 149 in run
  File "/usr/local/lib/python3/dist-packages/UM/JobQueue.py", line 125 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f2fcf4ba700 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 293 in wait
  File "/usr/lib/python3.5/threading.py", line 424 in ac2017-06-16 11:53:07,246 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] This program is distributed in the hope that it will be useful,
quire
  File "/usr/local/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/local/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.5/thre2017-06-16 11:53:07,246 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
ading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f2fe0129700 (most recent call first):
  File "/usr/local/lib/python3/dist-packages/cura/CuraApplication.py", line 661 in run
  File "/usr/local/bin/cura", line 73 in <module>
2017-06-16 11:53:07,247 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
Segmentation fault (core dumped)

nallath, 2017-06-19:
@Ghostkeeper; Maybe something caused by the ironing?

smartavionics, 2017-06-19:
It's only happened the once, not sure what triggered it.

Ghostkeeper, 2017-06-29:
It's not the ironing, I'm sure. That only made changes to CuraEngine, apart from the settings in fdmprinter.

smartavionics, 2017-06-29:
Well, it was a one-off event so let's close this due to lack of evidence...

BTW, I tried the ironing on a PLA  part and the surface quality was exceptionally good. Took forever to do even a small area but it did look nice.

closed 2017-06-29 by smartavionics