Ultimaker/Cura #1984: Cura 2.5 and 2.6 crash on start

lmanstl, 2017-06-17:
App version: 2.5.0 and 2.6.0
Platform: Windows

I have been using cura for a few days without issues. but, When I clicked the drop down arrow for the quality profiles it cause cura to crash and it has not been able to open since. I have used the quality profiles before without issue so I don't know what caused this.

Log: [cura.txt](https://github.com/Ultimaker/Cura/files/1082261/cura.txt)
The log was located in C:\Users\***\AppData\Local\cura not C:\Users\***\AppData\Roaming\cura like the readme says

Ghostkeeper, 2017-06-29:
What's this "IIIP" machine?

I can't reproduce this. I use the quality profile drop-down all the time and it works fine.

Perhaps it would help to zip up that entire folder at `C:\Users\Username\AppData\Local\cura`.

The readme is updated for 2.6, where it gets stored in Roaming instead of Local. If you're still using 2.5 you would technically have to use the version of the source code at that version: https://github.com/Ultimaker/Cura/tree/2.5.0

lmanstl, 2017-06-29:
I use Cura with my Monoprice Maker Ultimate and IIIP is the company that actually produces the printer, so that is what I named the custom printer profile that I set up. 

I also can't reproduce this on any other computer I own. I have Cura installed on a Mac and another windows machine with almost the same settings and I can't get this to happen again.

The really annoying thing is that this happend on personal machine which is the one that I use for everything.

It chashes everytime I open it no matter how many times I reboot or reinstall the program. I also just tried installing 2.6.1 and that also crashes.

Here is the whole folder
From C:\Users\Username\AppData\Roaming : 
[Cura(roaming).zip](https://github.com/Ultimaker/Cura/files/1112435/Cura.roaming.zip)
From C:\Users\Username\AppData\Local : 
[cura(local).zip](https://github.com/Ultimaker/Cura/files/1112437/cura.local.zip)

Thanks for your help!

Ghostkeeper, 2017-07-04:
It doesn't crash when I load them :(

It doesn't load the files either though. Looking at the zip, it seems that the machine instance got missing after upgrading to 2.6. I'd expect it to give an error in the log about this (and there is) but not that it crashes...

ianpaschal, 2018-05-22:
10.5 months old, never reproducible, won't patch 2.5/2.6 anyway, closing.