Ultimaker/Cura #1985: Cura 2.6 crashes when select multiextruder custom printer

cadshef, 2017-06-17:
Version: 2.6.0-BETA
Platform: Windows-7-6.1.7601-SP1
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\BuildVolume.py", line 491, in _onStackChanged
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\BuildVolume.py", line 555, in _updateDisallowedAreas
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\Settings\ExtruderManager.py", line 459, in getUsedExtruderStacks
KeyError: '1'

-------------------------
Thread 0x0000099c (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 150 in _storeStderrToLogThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00000f94 (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 142 in _storeOutputToLogThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00000304 (most recent call first):
  File "C:\Program Files\Cura\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x0000016c (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00000630 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x000007ac (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 297 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1153 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00000cc0 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 297 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x000005a4 (most recent call first):
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00000890 (most recent call first):
  File "C:\Program Files\Cura\plugins\RemovableDriveOutputDevice\WindowsRemovableDrivePlugin.py", line 63 in checkRemovableDrives
  File "C:\Program Files\Cura\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 56 in _updateThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00000d18 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 297 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1153 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00000bd4 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 297 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00000d30 (most recent call first):
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00000ac4 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 297 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00000fb0 (most recent call first):
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Current thread 0x00000310 (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "D:/2.6/build/inst/bin/cura_app.py", line 44 in exceptHook
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 322 in _onActiveExtruderStackChanged
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\Settings\ExtruderManager.py", line 514 in __globalContainerStackChanged
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 315 in __performEmit
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 212 in emit
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Application.py", line 152 in setGlobalContainerStack
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 346 in setActiveMachine
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 659 in run
  File "D:/2.6/build/inst/bin/cura_app.py", line 70 in <module>
  File "D:\env\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "D:\env\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

maukcc, 2017-06-20:
maybe this has something to do with : #1948
So uncheck those settings and see if that helps

Ghostkeeper, 2017-06-29:
I don't think it has anything to do with #1948.

This error occurs when it can't load some of the extruders. It's hard to tell from this log why that happened since the error occurred earlier. It says that this happens in the build volume since the updating of the disallowed areas is the first thing that usually happens when you switch to the machine.

nallath, 2017-07-27:
This is probably already fixed. We had a few fixes in 2.6.0 regarding this.

closed 2017-07-27 by nallath