Ultimaker/Cura #1986: Problems with circular/concentric patterns/infill

ex-nerd, 2017-06-17:
Cura 2.5, MacOS.

I recently tried to use the circular top/bottom pattern, but it failed to bridge properly on the top because it assumes there is infill or some other support structure to hold up the very center of the top pattern.

Replication steps:
* Load a cylinder into Cura
* Select concentric infill (some low percentage like 5-10%)
* Select circular top/bottom pattern
* Make sure you will render at least 3 top layers
* View the layers the Cura will render

Two things I've noticed:
* Concentric infill likes to print inner circles suspended in mid-air (with no support below them).
* The first layer of the top that will be printed (e.g. third from the very top) is at risk of being suspended in mid-air.

My thoughts on how to fix:
* Concentric infill needs some sort of vertical support going through the center of the concentric circles, so that there is a platform upon which to build each of the concentric "spheres".
* When using circular top/bottom patterns, the first printer layer of the top layers should be optimized for bridging (probably "lines") so that there is a more solid base upon which to build the subsequent circular patterns.

Ghostkeeper, 2017-06-29:
This is sort of expected. Cura has no explicit areas yet that it marks for bridging. This has been a long requested feature. In fact, internally this has ticket number CURA-67. Currently we're at ticket CURA-3980 and that no. 67 is still open!

BagelOrb, 2017-07-14:
This request is about the top skin being bridged over the supporting infill.

The solution is to just not use concentric top/bottom pattern or infill pattern. Those patterns just don't work very well.

Perhaps if you do it because you like the look on the top you can be helped by my new feature Roofing: which separates the top most skin from the rest of the top/bottom skinfill.

closed 2017-07-14 by BagelOrb