Ultimaker/Cura #1987: 1mm wide lines print 8mm wide

abetterway, 2017-06-18:
1mm wide lines print 8mm wide...
They show correctly in the viewer but print extra wide for some reason.
How do I get to the log to show you what is happening.

nallath, 2017-06-19:
Could you provide us with some more reproduction steps (For intstance, a project file where this happens)

abetterway, 2017-06-25:
Sorry, I'll need to set up a google drive to link to

Ghostkeeper, 2017-06-29:
You can put these files in a .zip (or even rename them to .zip) and drag them in the comment box here.

abetterway, 2017-06-30:
Everything seemed to be working until I got it on the printer and saw the layers.
These are really .gcode and .stl respectively.  I assume it's operator error but I don't see what happened.
Please take a look if you would.
Thanks
[z_height_set_180mm_.zip.zip]
[test_180 v2_was_stl.zip](https://github.com/Ultimaker/Cura/files/1113955/test_180.v2_was_stl.zip)
(https://github.com/Ultimaker/Cura/files/1113928/z_height_set_180mm_.zip.zip)

Ghostkeeper, 2017-07-05:
Those ZIP files are corrupted, according to my archiver :(

fieldOfView, 2017-07-05:
That's because they are .gcode and .stl files instead of .zip files (as mentioned in the comment)

Ghostkeeper, 2017-07-05:
Good point. I didn't read properly, sorry.

It looks suspiciously like you've accidentally set the Horizontal Expansion setting to something large, like 3.5mm. Did you change the Horizontal Expansion?

BagelOrb, 2017-07-14:
Closing due to inactivity.

closed 2017-07-14 by BagelOrb