Ultimaker/Cura #1989: 2.6 beta: Crash after dragging in and rotating simple object

ewanold, 2017-06-18:
Version: 2.6.0-BETA
Platform: Windows-10-10.0.14393
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Qt\Bindings\MainWindow.py", line 131, in mouseMoveEvent
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Qt\QtMouseDevice.py", line 30, in handleEvent
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 212, in emit
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 315, in __performEmit
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Controller.py", line 255, in event
  File "C:\Program Files\Cura 2.6\plugins\Tools\RotateTool\RotateTool.py", line 116, in event
    drag_start = (self.getDragStart() - handle_position).normalized()
TypeError: unsupported operand type(s) for -: 'NoneType' and 'Vector'

-------------------------
Thread 0x000026c0 (most recent call first):
  File "C:\Program Files\Cura 2.6\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 58 in _updateThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x0000235c (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 297 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1153 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x000027b0 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 297 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00002a40 (most recent call first):
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x000010c0 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 297 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00002708 (most recent call first):
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x000024b4 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00002804 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x0000279c (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00002850 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Current thread 0x00001f50 (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "D:/2.6/build/inst/bin/cura_app.py", line 44 in exceptHook
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 659 in run
  File "D:/2.6/build/inst/bin/cura_app.py", line 70 in <module>
  File "D:\env\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "D:\env\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

Ghostkeeper, 2017-06-29:
It's not happening when I try it. Works fine.

Does it happen for all models? Perhaps when you drag on a specific part of the model?

ewanold, 2017-06-29:
I don't remember that particular model any more. All I can say: the beta crashes from time to time, which is maybe around every 10 times, but nothing I could reproduce. Maybe it needs loading a certain amount of different models without restarting between. One of these nasty bugs I encounter myself every now and then.

I will switch to 2.6.1 the next days and observe the matter :-) If it helps in some way, close the bug for now.

BagelOrb, 2017-07-14:
I guess this bug was solved because you didn't respond any more.

Ghostkeeper, 2017-07-20:
It's not. They found this in QA as well but it's hard to reproduce.