Ultimaker/Cura #1990: 2.6. beta: German translation and delete shortcut

jstrobel, 2017-06-18:
On the german translation there are some snippets missing in german. See attachted screenshot.

And why isn't backspace not working like before and was changed to forward delete? I think this isn't intuitive, but this is just my opinion.

[image]

Ghostkeeper, 2017-06-29:
Backspace was added as an alternative to delete because many Macintosh computers and keyboards don't have a delete key. We did this quite a long time ago, I think in October 2015.

Ghostkeeper, 2017-06-29:
I looked into the translations, the actual issue presented here.

The translations weren't correct in the beta because these strings changed between 2.5 and 2.6 (or were added in the case of Arrange All Models). The 2.6 beta was made before we had the translations for the 2.6 release. It should be correct in the actual 2.6.0 and 2.6.1.

jstrobel, 2017-06-29:
@Ghostkeeper But it is not a backspace which it was before and 100% fine. It is now a **forward delete**.

closed 2017-06-30 by Ghostkeeper