Ultimaker/Cura #1996: Build Plate Adhesion - Skirt Settings Ignored

JHBP2, 2017-06-20:
I set Build Plate Adhesion Type to None, but it still creates a skirt. The Skirt Distance setting is also ignored. It is stuck on 0, in contact with the first layer. Tried switching profiles, exporting / importing, turning off and on. Still ignores it.

nallath, 2017-06-20:
Uh, what kind of machine do you have? If a machine has one extruder, it shouldn't have the extruder tab.

JHBP2, 2017-06-21:
The machine only has one extruder.

JHBP2, 2017-06-21:
I also noticed that even though the skirt settings are displaying
incorrectly, and rendering incorrectly in layer view, turning it off
actually works in the gcode...?!? I am not sure if the distance setting
would do the same i.e. display as 0, render as 0, but then work correctly
and leave the proper gap in a real print.

JHBP2, 2017-06-22:
Nope... The Skirt Distance setting is being ignored in the gcode. This "Extruder 1" could be the problem... It does not seem to pick up the settings, and may be overriding them...?

[image]

fieldOfView, 2017-06-22:
@JHBP2: you're using "Custom FDM Printer", right? In that case this is fixed with #2009

You can test this fix:
* Download this file: https://raw.githubusercontent.com/fieldOfView/Cura/adc547fa36b3edc288369543d1c97c648f6f22c5/plugins/CuraEngineBackend/StartSliceJob.py
* Replace the file on your computer that is located in [cura install location]/plugins/CuraEngineBackend (eg: ```C:\Program Files\Cura 2.6\plugins\CuraEngineBackend\StartSliceJob.py``` on windows) 
* Restart Cura and test if it works now.

JHBP2, 2017-06-22:
@fieldOfView Thanks! Yes, I am using "Custom FDM Printer"... This fix looks good so far in Layer view, but I'll have to wait until my current print is finished tomorrow before I can do a real print test. Thanks again 👍

[image]

fieldOfView, 2017-06-22:
If it is good in layerview, it is good in the print.

- knocks on wood -

JHBP2, 2017-06-23:
@fieldOfView Success! Skirt settings work with this fix!

[image]

[image]

closed 2017-06-23 by nallath