Ultimaker/Cura #1997: G-code is generated at Z0 even though object is placed at a higher level

MikeBurger001, 2017-06-20:
When setting the Z height of a model at a value other than zero Z, the g-code generated starts at Z0
When doing the preview as "solid", the object appears above the build plate, but when viewing using "layers" the g-code is generated from Z0m level.
I wanted to re-use a raft previously printed, and therefore start my model at 1.55mm above the build plate.
That does not seem possible :(

nallath, 2017-06-21:
Nope, sorry. Empty layers are removed.

Ghostkeeper, 2017-06-30:
Workaround is to place a tiny cube in a corner of your build plate that's at least as high as your raft.

BagelOrb, 2017-07-14:
We're not going to implement something to help with such problems generally.

closed 2017-07-14 by BagelOrb