Ultimaker/Cura #1998: Source code ultimaker 3 firmware

14bitVoid, 2017-06-20:
Is there anywhere I can find the firmware for the ultimaker 3?

I'd like to try to make the printer handle arcs better, because its not working very well for me right now.

nallath, 2017-06-21:
The firmware is not released yet.

closed 2017-06-21 by nallath

14bitVoid, 2017-06-21:
Yes it is. It's right here: https://ultimaker.com/en/resources/23129-updating-firmware

nallath, 2017-06-22:
Sorry, i meant the source of it is not released.

14bitVoid, 2017-06-22:
Oh... isn't that based on marlin? You kinda have to release the source code.

fieldOfView, 2017-06-22:
See what @daid has to say about that:
https://ultimaker.com/en/community/23254-inside-the-ultimaker-3-day-1-gcode?page=3#reply-162578

daid, 2017-06-23:
If you are using G2/G3 arcs, that's generally a bad idea. As those completely block communication with Marlin, and I almost decided to remove support for those. As blocking communication is something we normally prevent to ensure proper operation with Marlin and the rest of the system.