Ultimaker/Cura #2001: Anti-overhang meshes are ignored if they go out of bounds

TheAMM, 2017-06-21:
This makes excluding generated supports near build plate edges/corners bothersome, as the user will have to make sure the A-O mesh won't go out of bounds *or* use several meshes near corners.

As meshes marked anti-overhang are not printed, they should not be ignored for going out of bounds.

**Example images** *(although not from a real use-case)*
Functioning support exclusion
[image]
Anti-overhang mesh ignored for going out of bounds
[image]

nallath, 2017-06-21:
For our reference CURA-3951

I tried to make a quick fix for this (as i thought this was "just" a front-end issue), but it seems the engine also needs a bit of work to fix this.

Ghostkeeper, 2017-06-30:
The engine typically doesn't take these bounds into account at all, right? Only to find the centre of the build plate sometimes.

Ghostkeeper, 2017-06-30:
Fixed this here: https://github.com/Ultimaker/Cura/commit/71c146f4379c24af2e05ffadd690c3b24e5ed4eb

closed 2017-06-30 by Ghostkeeper