Ultimaker/Cura #2003: Initial build plate temp&build plate temp..

muhendis80, 2017-06-21:
I made
Build plate temperature 0 C
Build plate temperature Initial layer 100 C

it is always working at 100 C, but it did not cold down temperature to zero after initial layer made

Ghostkeeper, 2017-07-02:
Indeed, this reproduces. I suspect because 0 degrees is treated as a special case where it would simply not be necessary to output any temperatures (until first layer bed temperature was introduced as a feature).

Workaround for now is to put the ordinary bed temperature to 1 degree. I'm filing a bug about this though.

Ghostkeeper, 2017-10-02:
You're going to have to keep using the workaround. Our project manager just removed this issue from our planning for being older than 12 weeks.

Ghostkeeper, 2017-11-06:
This one is earlier, but let's treat it as a duplicate of https://github.com/Ultimaker/Cura/issues/2674

closed 2017-11-06 by Ghostkeeper