Ultimaker/Cura #2004: Digits of Extruder axis output

machinekoder, 2017-06-21:
Is it possible to change the number of digits on the extruder axis output? When I calculate the cross section of the lines I can see that there quite some rounding errors for very short and thin lines.

nallath, 2017-06-21:
The rounding that I know of is at micron level. @BagelOrb or @Ghostkeeper can tell you more about the exact mechanisms.

machinekoder, 2017-06-21:
@nallath Yes, most lines are fine. However, sometimes short lines in the micron length can have quite high variations.

Ghostkeeper, 2017-07-02:
There's no setting for this. The accuracy of the E axis is actually down to 10nm, by the way, or 10nm^3 if you're working with a volumetric g-code flavour.

This is where the accuracy of the E axis gets determined: https://github.com/Ultimaker/CuraEngine/blob/6d69852d084e95cdfe9934a0e4e42e739cbe8c4e/src/gcodeExport.cpp#L686
If you are able to compile CuraEngine, I suggest you make the change there. If not, I'm willing to compile a drop-in replacement for Linux with more accuracy there.

I see no reason to make this an actual setting though. 10nm should be way more accurate than any extruder can handle, except some very specialised ones perhaps. Making it more accurate only serves to make the g-codes bigger and the slicing slower.

machinekoder, 2017-07-03:
@Ghostkeeper Thank you. I have to look into volumetric GCode then.

What I'm working on is [velocity extrusion](http://machinekoder.com/machinekit-and-cura/) for which I create the [NGCWriter plugin](https://github.com/machinekoder/NGCWriter). In my code, I convert the E axis value back to the line cross section. In my output GCode, I noticed very large tolerances in the calculated cross section which leads me to the conclusion that there must be some big rounding errors either in my plugin or CuraEngine.

Since Cura uses a single line thickness, I was able to correct the GCode afterward by replacing all cross section value with the intended one. However, the resulting quality difference is probably minimal.

I will take a closer look at the problem and see if I can provide some examples.

BagelOrb, 2017-07-03:
Line widths are actually modulated by neighboring lines. If you want Cura to use a single line thickness you should disable the setting Compensate Wall Overlaps.

The Compensate Wall Overlaps feature does have some limitations. It seems it is not extruding enough when there's a lot of small line segments really close to each other. However, I think this effect is way smaller than the effect of compensating for the overlap itself.

The feature kicks in where the walls got close to themselves - that is: when your model has thin pieces and where a layer has sharp angles.

BagelOrb, 2017-07-14:
I suppose this question is answered.

closed 2017-07-14 by BagelOrb

machinekoder, 2017-07-14:
Okay, that's what I might see in my output.