Ultimaker/Cura #2007: 2.6 Filament Diameter and Nozzle Temp Ignored

DDeGonge, 2017-06-21:
I'm using Cura 2.6 on Windows, this is a clean install with no previous profiles. My machine is set to custom with 1.75mm filament. I have created a filament profile as well with 1.72mm as the mean diameter and 210*C printing temperature. I have tried using both versions of Marlin (standard and volumetric), but standard always under-extrudes and volumetric always over-extrudes.

When viewing the full list of slicing settings, I am able to see the proper filament diameter and temperature settings. However, when the model is sliced, there are 2 problems.

1. It always uses a print temperature of 200*C. I have tried using different baked in profiles as well as adjusting the value in the slicing settings by hand, but nothing seems to change the generated gcode print temperature.
2. It always slices assuming filament is 2.85mm diameter, but the preview estimation stats are correct. For example, a solid 2cm cube slices and when it is complete, the stats in the bottom right say it will use 2.2m of filament (I forget the actual value but it is close to 2.2). This is correct for 1.75mm filament. However, when importing the generated gcode into repetier host, it shows that only 0.8m of filament are actually being used. When printing, it shows that in fact 0.8m is accurate as the print is severely under-extruding. This is using the standard Marlin gcode flavor. Using volumetric seems to do the opposite and results in severe over-extrusion.

Increasing the flow multiplier in Cura to 270% fixes the problem and prints nicely, but this is a bit hacky and doesn't address the cause. Additionally, I still can not figure out any way to adjust the sliced print temperature.

Let me know if you need more info!

-Andrew

fieldOfView, 2017-06-21:
What printer definition do you use? One of the predefined ones, or a Custom FDM Printer?

DDeGonge, 2017-06-21:
Custom FDM

fieldOfView, 2017-06-21:
Both issues can be checked in the generated gcode. Could you post it somewhere?

The print temperature is set near the top of the gcode file. There should be a line that says eg ```M104 S209``` if you set the temperature to 209 degrees.

The total amount of extruded material can be check by scrolling to the bottom of the gcode file and checking the last G1 before the end gcode. The E value shows the last "position" in mm.

fieldOfView, 2017-06-21:
Could you try lying to Cura and setting the number of extruders to 2?

DDeGonge, 2017-06-21:
[Here is the gcode for a test part](https://drive.google.com/file/d/0BzkIVnYJoiSgR20wNERQSGNPRzQ/view?usp=sharing)

I re-installed cura from scratch to try to narrow down possible causes. It seems the print temperature issue has to do with using the "default printing temperature" rather than "printing temperature". Using the latter works. Not sure the point of the default since it seems to always be overwritten by this.

Extrusion volume is still incorrect. As you can see in the gcode, it says it uses 2.8m. This is also what Repetier sees. However, if this were to be printed, it would be under-extruding.

Cura still displays what should be the correct value in the GUI of 7.5m. It seems this estimation does use the filament diameter setting, but the slicing engine does not.

When switching gcode type to volumetric marlin, it seems to do the reverse of the above and the result is 18m being extruded. Once again, the Cura estimation is correct.

DDeGonge, 2017-06-21:
Setting extruders to 2 worked and both cura and repetier now display the proper 7.5m of extrusion

fieldOfView, 2017-06-21:
And for my reference, did you set the diameter only in the sidebar, or also in the machine settings?

DDeGonge, 2017-06-21:
Both machine settings and the side bar

DDeGonge, 2017-06-21:
Changing it back to 1 extruder now works correctly as well.

Also I did neglect to mention I use a nozzle of 0.3mm, rather than the default 0.4mm. Not sure if this is at all related though.

DDeGonge, 2017-06-21:
Actually, I take that back. It seems to just continue using the last settings used with the 2 extruder settings.

So changing filament diameter with 1 extruder has no effect. I can also confirm changing nozzle size with 1 extruder has no effect. Changing nozzle size and filament diameter with 2+ extruders does work.

fieldOfView, 2017-06-21:
Thanks for reporting. I have made a fix here:
https://github.com/Ultimaker/Cura/pull/2009

You can actually help testing this fix:
* Download this file: https://raw.githubusercontent.com/fieldOfView/Cura/adc547fa36b3edc288369543d1c97c648f6f22c5/plugins/CuraEngineBackend/StartSliceJob.py
* Replace the file on your computer that is located in [cura install location]/plugins/CuraEngineBackend (eg: ```C:\Program Files\Cura 2.6\plugins\CuraEngineBackend\StartSliceJob.py``` on windows) 
* Restart Cura and test if it works now.

DDeGonge, 2017-06-21:
New file works!

Only thing worth noting is that changing the filament diameter in the printer settings does not hold if you edit it. Rather, it just inherits whatever the current filament diameter is set to in the slicing settings. Honestly, the filament diameter should probably be removed entirely from printer settings and just left to the slicing/material profile.

fieldOfView, 2017-06-21:
Heh, you know how much sh*t we get for not having the nozzle size in that list?

The material diameter is not visible by default. You made it visible. There is a usecase for having it visible, so it will probably not be removed any time soon.

The machine settings dialog could be more informative, telling you the value is not used when it is overridden in the sidebar or elsewhere

DDeGonge, 2017-06-21:
Good point, it does seem the machine settings dialog got overlooked while all the slicing settings were getting the royal dialogue treatment haa. Perhaps a bit more info with those printer settings could go a long way to avoid confusion.

Anyway thanks for finding a fix so quickly! Now I can officially make the jump from Repetier/Cura to Cura standadlone! 😄

nallath, 2017-06-22:
The pull request form fieldOfView is merged. So this is fixed for 2.6.1

closed 2017-07-02 by Ghostkeeper