Ultimaker/Cura #2008: 2.6 release: Settings changes from profile ignored until profile updated

giseburt, 2017-06-21:
I haven't done an extensive search as to what all settings are affected, but in the least it's the Infill Density and Infill Pattern.

If I change the infill density or the infill pattern, it will trigger a reslice, but the slice will be with the old settings. It appears to ignore the on-screen profile overrides and only use the settings from the profile. If I update the profile with the new overrides, it will trigger a reslice and the setting will be used correctly.

I'll update shortly with a video.

fieldOfView, 2017-06-21:
Are you by chance using a Custom FDM Printer with a single extruder? In that case please see #2007 and the fix in #2009.

You could try this fix:
https://github.com/Ultimaker/Cura/issues/2007#issuecomment-310144781

giseburt, 2017-06-21:
I am using a Custom FDM printer with one extruder. Sorry I failed to mention that.

I also looked and somehow missed that #2007 is the same issue. I'll close this one now since fixing that one will fix this one.

Thank you!
  -Rob

closed 2017-06-21 by giseburt