Ultimaker/Cura #2010: Nao aparece a mesa nem a peca a ser fatiada.

Guisao5, 2017-06-21:
Na linguagem em portugues nao aparece a mesa da maquina e nao e possivel carregar arquivos.

Ghostkeeper, 2017-07-02:
I've tested the Portuguese language in Cura myself. Just retesting to confirm:

* The printer list is properly filled in when you click "Adicionar Impressora". Adding a printer works fine.
* Loading a file works fine, and it starts to "Fatiando" right away.

In other words, I can't reproduce your problem.

Ghostkeeper, 2017-07-02:
Perhaps you can share a log file with us? I'd expect these problems to show up there.

ChrisTerBeke, 2017-09-28:
Closing since we cannot reproduce and no futher information has been provided. Please comment if this is still an issue for you.

closed 2017-09-28 by ChrisTerBeke