Ultimaker/Cura #2014: Shell, Bottom Pattern, Concentric Pattern is Almost Awesome!

JHBP2, 2017-06-22:
The Concentric Top and Bottom pattern is a beautiful feature, but it could be AWESOME! At the moment, one "seam" line is created that radiates out from the middle. It looks like it is the spot on the circle where the diameter changes. Is there a way to make this pattern behave more like a Spiral, so there is no seam at all? It would look awesome! Or perhaps it could be a new pattern called "Spiral" if folks like the current one too...?

[image]

JHBP2, 2017-06-22:
This angle shows it better:
[image]

BagelOrb, 2017-06-22:
This is known as Fermat spiral infill.
We aren't actively developing it. It's quite hard to get it to work for all shapes.

JHBP2, 2017-06-23:
That's OK. I appreciate you considering it, and the new Cura with the new interface is still AWESOME! 😁👍

BagelOrb, 2017-06-23:
Haha, thanks! :D

closed 2018-11-22 by Appesteijn

BagelOrb, 2018-11-22:
The connect skin polygons in combination with concentric infill does something very similar to what you are asking.

Amritpal363, 2025-05-18:
Good morning 🙏🌹 india punjabi city code 143604

[image]

Amritpal363, 2025-05-18:
🙏🙏🙏😔😔😔

Amritpal363, 2025-05-26:
[image]