Ultimaker/Cura #2017: 2.6 win64 doesn`t launch

digiTALker, 2017-06-22:
Installed correctly with and without Administrator rights.
But not starting. No any errors just silence.

fieldOfView, 2017-06-22:
C:\Users\[username]\AppData\Roaming\cura\2.6\cura.log

digiTALker, 2017-06-23:
Wow file takes 65mb and seems adds 3mb after each start.
https://yadi.sk/d/2-PciT8P3KPBE8

fieldOfView, 2017-06-23:
There's a problem reading the ```custom_1mm+abs+крупные+грубые+одиночки+без+отка.inst.cfg```, probably because the urlencoded version of the filename is longer than 256 characters (```C:\\Users\\Надёжный\\AppData\\Roaming\\cura\\2.6\\./quality\\.\\custom_1mm+abs+%D0%BA%D1%80%D1%83%D0%BF%D0%BD%D1%8B%D0%B5+%D0%B3%D1%80%D1%83%D0%B1%D1%8B%D0%B5+%D0%BE%D0%B4%D0%B8%D0%BD%D0%BE%D1%87%D0%BA%D0%B8+%D0%B1%D0%B5%D0%B7+%D0%BE%D1%82%D0%BA%D0%B0.inst.cfg```). Cura is going to have to be more resilient to long names like this.

You can probably fix it by making the filenames in C:\Users\Надёжный\AppData\Roaming\cura\2.6\quality\ shorter.

The log file contains all the logs back to august 30 2016, which is why it weighs 65 MB. I can't see that 3mb is added on every start; it seems only 8.3 kb is added.

Ghostkeeper, 2017-07-03:
We have this for machine names, here: https://github.com/Ultimaker/Cura/blob/master/cura/Settings/MachineNameValidator.py#L31
Quite exhaustive. It actually checks the file system to see how long the file name is allowed to be. All non-ANSI characters count as 12, to simplify the UTF-8 that we use.

But I think we lack this exhaustive check for profile names.

Ghostkeeper, 2018-02-28:
Our project manager removed this from our planning. Not for now, he said. Maybe we'll address this later but don't count on it for now.
Using shorter names or Latin characters will prevent this from happening, so you'll have to stick to that for now.

closed 2023-06-24 by github-actions[bot]