Ultimaker/Cura #2019: Cura 15.04.06 crashed/reset on Windows 10

avipars, 2017-06-23:
I was using Cura for a little less than a year on my Geeetech Prusa i3, this is the first time it crashed and it reset from scratch, I loaded in gcode and a cura profile, but still didn't work....

What are my options, is there a way to recover from this?

fieldOfView, 2017-06-23:
Cura 15.04 is no longer supported.

closed 2017-06-23 by nallath

avipars, 2017-06-23:
Which version still supports 3rd party printers ?

nallath, 2017-06-23:
Cura 2 also has a pretty big list of third party printers.

avipars, 2017-06-23:
Will it still support Marlin and prusa i3 clones?

nallath, 2017-06-23:
Probably. You would have to check if your specific combination works.

Ghostkeeper, 2017-07-03:
We have definitions for the Prusa i3. Not for this Geeetech clone specifically though, but some minor tuning should clear that right up.

Generally, Cura assumes you're using Marlin or something that quacks like Marlin. At least when connecting via USB...