Ultimaker/Cura #2020: Feature request: Change led on UM3 through Cura

Dim3nsioneer, 2017-06-23:
With the video feed in Cura you can check what your UM3 is doing. As some people might be doing this pretty remote (via VPN), it would be nice to have a possibility in Cura to set the LEDs on the UM3. While you need the LEDs on to see what the printer is doing you might want to have them dimmed or off the major time. So in addition to the pause and cancel buttons in Cura I suggest a setting for the LED brightness on the interface page during print.

nallath, 2017-06-23:
What would be good position to put the control for that do you think?

Dim3nsioneer, 2017-06-23:
Somewhere left of the cancel button so people realise it's something they can change? Alternatively close to the preheat control.

Dim3nsioneer, 2017-06-23:
something maybe like this:

-----------------
| - LEDs: 60% + |
-----------------

Ghostkeeper, 2017-10-10:
Would've been nice to have, but our project manager just removed this from our planning because it was older than 12 weeks.

closed 2018-11-22 by Appesteijn