Ultimaker/Cura #2022: Failing pytests

thopiekar, 2017-06-24:
Just wanted to update my packaging files, because there is the new PluginManager and found in the meantime failing tests:

https://launchpadlibrarian.net/325247492/buildlog_ubuntu-xenial-amd64.cura_1%3A2.5.99-master~201706231602~rev2246~pkg154~ubuntu16.04.1_BUILDING.txt.gz

Is that known? I thought we are using our server for tests now and we don't merge anything before the tests have passed :thinking:

awhiemstra, 2017-06-27:
It looks like something went really wrong with our Jenkins CI master. I am not yet sure what happened but we are investigating right now.

thopiekar, 2017-06-27:
There the latest log, but I don't expect any changes:
https://launchpadlibrarian.net/325739043/buildlog_ubuntu-zesty-amd64.cura_1%3A2.5.99-master~201706271147~rev2261~pkg156~ubuntu17.04.1_BUILDING.txt.gz

thopiekar, 2017-06-29:
Also errors in 2.6:
https://launchpadlibrarian.net/326008720/buildlog_ubuntu-zesty-amd64.cura_1%3A2.6.1~201706290710~rev2249~pkg157~ubuntu17.04.1_BUILDING.txt.gz

But they look different, so could be that it is using the 2.5 Uranium. (haven't noticed that I haven't updated my PPA for a while :roll_eyes: )

thopiekar, 2017-06-29:
Don't worry. It were the old deps. All fine except of something else, which is not related to cura itself.
https://launchpadlibrarian.net/326010597/buildlog_ubuntu-yakkety-amd64.cura_1%3A2.6.1~201706290730~rev2249~pkg157~ubuntu16.10.1_BUILDING.txt.gz

thopiekar, 2017-06-30:
Works again: https://launchpadlibrarian.net/326229408/buildlog_ubuntu-artful-amd64.cura_1%3A2.6.99-master~201706301202~rev2281~pkg159~ubuntu17.10.1_BUILDING.txt.gz

closed 2017-06-30 by thopiekar