Ultimaker/Cura #2023: v2.6.0 not importing materials

kr15uk, 2017-06-24:
Ladies and gents,

Exported materials out of Cura v2.6.0 on my Mac (10.12.5) and imported on Windows machine (10 x64) Cura v2.6.0.
Getting message - sucessfuly imported but nothing apears.
Tried to quit and reopen - with no success.

I'm open for testing if required.

Kris

fieldOfView, 2017-06-24:
What printer do you use? Does the printer you added on the Mac use the same diameter filament as the printer on the Windows installation?

Can you post the material files somewhere so we can have a look?

kr15uk, 2017-06-24:
Same machine + same impoted config profile.

Materials - https://drive.google.com/drive/folders/0B-4wTtCQEkGecTh5bDN0UUFtUlU?usp=sharing

Thanks.

Ghostkeeper, 2017-07-03:
I tried adding that BQ-Green profile. It works for me, but only if I am using one of the printers marked as supported in that material, such as the Cartesio.

nmaraujo, 2017-07-04:
BQ Witbox2 profiles don't load too

jackha, 2018-07-26:
If there are any problems regarding this in the newest version (Cura 3.4.1) then I'd be happy to reopen.

closed 2018-07-26 by jackha