Ultimaker/Cura #2024: Cura 2.6.0 OSX not recognizing when removable drive is no longer available

LevelheadCA, 2017-06-24:
- When an SD card is ejected (from the Finder or the Cura 'eject' button) 2.6 Cura still shows the "Save to removable drive" button. Clicking the button does nothing.  If you re-mount the SD card clicking "Save to removable drive" is non-functional (does nothing when clicked). 

- If you switch to 'Save to file' from "Save to removable drive"  and then insert an SD card, a checkmark appears next to "Save to removable drive" but only "Save to file" appears in the UI.

- on launch Cura does recognize an SD card and "Save to removable drive" functions properly. However once ejected, "Save to removable drive" remains in the UI and is non-functional.

PapaG-Design, 2017-06-29:
Having the same issue.  Once a file is saved to SD card I have to restart Cura to save again to SD Card

nallath, 2017-06-29:
What version of OSX are you using?

LevelheadCA, 2017-06-29:
I am on MacOS Sierra 10.12.5 on a 27" iMac (booting off external thunderbolt drives if that helps).  I have tried a clean install of 2.6.0 and get the same issue.  One thing I noticed during the fresh install was that the first time I launched Cura 2.6.0, the icon bounced for a second and then nothing happened. The second time I launched Cura, It crashed and I received an error report (will post if I can find it). For the third and all following launches, Cura started properly.

supersoulman, 2017-06-30:
same issue here. Running OS 10.12.5 with CURA 2.6.1

When I save a file to an SD card I can't save to an SD again unless i restart CURA. I don't remember having this issue in CURA 2.5-2.6

cmatthes, 2017-07-08:
Same issue running macOS 10.11.6 with Cura 2.6.1 . 

Need to restart Cura after I save a file to USB.

yyh1002, 2017-07-27:
2.6.0, 2.6.1, 2.6.2 all have the same issue for me while 2.5 doesn't.
Basically a deal breaker. 
Running Windows 10.

ericdlaspe, 2017-08-08:
Same problem here on macOS Sierra 10.12.6.  I have to restart Cura every time I want to reinsert an SD card on Cura 2.6.2.  I did not have the same problem with 2.5.

ericdlaspe, 2017-08-08:
As this is a deal breaker for some of you, it's worth mentioning this does not seem to be a problem in Cura 2.7 (beta).

Ghostkeeper, 2017-08-23:
We had this issue in our internal tracker as well and then marked it as "Cannot reproduce" with the note that it was very likely fixed by this commit: 23092ac1325fc95b8872b26baa1e40a1b97232eb

closed 2017-11-06 by Ghostkeeper

ChrisTerBeke, 2017-11-06:
Not having this issue on Mac at least.