Ultimaker/Cura #2025: Machine settings taken from 2.5 are wrong (Diameter), won't update unless changed in print setup.

Cyberman-tM, 2017-06-25:
Cura 2.6.0 , Windows 7, Wanhao i3, custom profiles, had 2.5.something before

Two problems:

1) When I installed Cura 2.6 , the material diameter was changed to 2.85 - the rest of the settings apparently was taken over correctly?
I assume this is directly related to https://github.com/Ultimaker/Cura/pull/2009

2) When I change the material diameter in the printer machine settings it gets ignored in favor of whatever is in the print setup (i.e. sidebar).
Changing the nozzle size does work, however, probably because it's not in the sidebar.
Is that intentional? If so, I think there should be a warning somewhere - I wasn't even aware that there is a diameter setting in the sidebar, I've hidden everything but temperature in the material section.

See also: https://ultimaker.com/en/community/49977-editing-machine-settings-cura-260-beta
(Without that forum thread I'd never have found the reason...)

rpgrunwald, 2017-06-28:
Confirmed in 2.6.1  Win-64. 
I upgraded from 2.5 and let it uninstall 2.5.  I used all default installes. 
I also have custom print and material profiles.  They came over fine.
Diameter would not latch in printer profile unless I changed it in the sidebar from 2.85 to 1.75.  Then it seemed to latch in the printer profile.

Ghostkeeper, 2017-07-03:
Hmm, I thought it would be fixed in 2.6.0 by this pull request: https://github.com/Ultimaker/Cura/pull/1890

johnboiles, 2017-07-11:
Confirmed in 2.6.2 on macOS

jwyse, 2017-07-19:
Same issue here.  NEW install of Cura 2.6.2 Win64 (unlike OP and other commenters, I did not upgrade from a previous version).

I'm using the Prusa i3 machine profile for my Monoprice Maker Select V2.  It seems to keep the setting once I change it in the sidebar for each profile, as rpgrunwald described.

rewthing, 2017-08-14:
Win64 2.6.2 here.  I upgraded from 2.4 and also see this issue.  I have three different printer profiles, and none of them will keep 1.75mm diameter when I attempt to set it.

thinkyhead, 2017-08-26:
*ping* I have also just found that the **Material Diameter** setting will not stick. Using Cura 2.6.2 on macOS.

nallath, 2017-08-28:
If all is well, this should be fixed in the 2.7 final release.

closed 2018-01-22 by ChrisTerBeke