Ultimaker/Cura #2026: Line width problem in new layer view

smartavionics, 2017-06-25:
I have noticed that the new layer view shows some pretty weird line widths. Here's an example:

[image]
The top layer has two walls close together and with wall overlap compensation enabled, the outer wall gets thinned. But look at the outer wall in the above image, the longish straight segments are all too thin and the corners are rather bumpy. The actual print looks absolutely fine so it must be a graphics issue. This is just one example, I could post lots of images of other walls whose widths (in the layer view) are silly.
My current theory is that the array of line widths sent to the UI are not synchronised with the vertices. So in ProcessSlicedLayersJob, if I add this line (after line 128) to rotate the array of line widths:

line_widths = numpy.roll(line_widths, -1);

I now get this:
[image]
Which is much more sensible and other wall lines that had weird widths now look good too so that rotation of the array is good for the walls. Unfortunately, it causes the skins and infill to be replaced with very thin lines as shown below and I don't know what to do next.

[image]
Could someone who understands this python + GL stuff please look into this as I don't want to waste loads of time working out a fix if someone who is familiar with the code can do the same in a fraction of the time?

smartavionics, 2017-06-26:
OK, I figured out a fix so that now all lines are showing with sensible widths but I would be the first to admit that I am thrashing around in the dark here so it could well be that the real problem is elsewhere and this change is just compensating for a bug elsewhere?

nallath, 2017-06-26:
@jackha You made the original implementation of this, what do you think?

closed 2017-06-26 by nallath