Ultimaker/Cura #2027: Material file GUID values, how to generate?

OutsourcedGuru, 2017-06-25:
I've attempted to export/edit/re-import a material file for a particular kind of filament but this appears to then replace the previous one I've clone from.  I'm guessing that the GUID, since I didn't edit that, could be used as a primary key perhaps.

If so, and it's necessary to generate a unique GUID for each custom material profile, do you know how we're supposed to do this?

nallath, 2017-06-26:
GUID can not be used as a primary key for a material. A GUID uniquely identifies a material, but a single material can have multiple interpetations (eg profiles).

Cloning indeed makes a profile that has the same GUID as the existing one. If you create a new material it will also have a new GUID.

fieldOfView, 2017-06-26:
You can use https://guidgenerator.com

Cura 2.6 also lets you "unlink" materials on the Materials preference pane, which also creates new guids.

OutsourcedGuru, 2017-06-26:
Awesome, thanks everyone.

closed 2017-06-26 by OutsourcedGuru