Ultimaker/Cura #2028: cora 2.6 unable to upgrade firmware

Pascal1573, 2017-06-25:
I'm Using a Ultimaker Original plus and i'm unable to upgrade the firmware.
"due to a communiaction error".
This works fine using earlier versions V2.5 and before.
(I also reported this using the 2.6 Beta)

I removed everything and reinstalled, without a result.
Returning to Cura 2.5.
[image]

nallath, 2017-06-26:
Could you post the logs?

ChrisRiddell, 2017-06-26:
I had this same issue on windows 10, worked fine on OSX though.

Pascal1573, 2017-06-27:
Sorry for the late responce I will upload the log asap. not able to at the moment. sorry!
as soon as i'm able to, i'm going te reinstall 2.6 and send the log!

By the way I use windows 10 pro (64bit) (ann. ed.)

Pascal1573, 2017-06-28:
Hi I installed cura2.6 and tried to update firmware.
the cura log is 2 days old, but the stderr and stdout logs are "from now".
It looks like the printer is detected at com4, but  then the update procedure tries to use com1.
I did see cersion 2.6.1 going to install that one now.

[stdout.zip](https://github.com/Ultimaker/Cura/files/1110189/stdout.zip)

nallath, 2017-06-29:
I don't see anything specific that could explain this in the logs. Weird. I'll ask testing if they also encountered the issue.

Pascal1573, 2017-06-29:
ok (?)
If I must test something just tell me and I'll do it.
With Cura 2.5 and before it works just fine even now.
It's the same USB cable the same PC the same UMO+.
When using cura 2.6, do I have to upgrade firmware? Or could I use cura 2.6(.1) and the firmware that I loaded with 2.5?

nallath, 2017-06-29:
We didn't change any code regarding the USB upgrading (or any USB related stuff) as far as I know. Could you try if uploading the firmware with the install custom firmware button works (and then upload the firmware from version 2.5)?

fieldOfView, 2017-06-29:
> When using cura 2.6, do I have to upgrade firmware? Or could I use cura 2.6(.1) and the firmware that I loaded with 2.5?

You should be able to run Cura 2.6 with the firmware that came with Cura 2.5 just fine.

Pascal1573, 2017-06-29:
nallath i did what you asked.
I used cura 2.6.1 to upload the firmware provided with cura 2.5
[image]
as you can see again unsuccessfull.
and here are the logs again.
[cura.zip](https://github.com/Ultimaker/Cura/files/1113499/cura.zip)

Equinox1st, 2017-06-30:
Hi there, i have the same Problem. Using Win10 64 Bit and the latest Cura 2.6.1. I'm not able to do the upgrade, it always stops with the communication Error mentioned above. But when i check the Version on my Ultimaker 2+ it tells me Firmware Version 2.6 14. Jun. With Cura 2.5 everything worked fine (It installed the 2.4 Firmware propperly)  I also deleted the Cura Folders in AppData Roaming // Local. No effect. If you want some logs... Where can i find them? :D

Pascal1573, 2017-06-30:
Hi!
The logs are in:
c:\users\<username>\appdata\cura

Equinox1st, 2017-06-30:
Thank you :) I'll post them on Monday

fieldOfView, 2017-07-01:
Note that for 2.6, the logs are in /roaming/cura/2.6, not in /local/cura

Equinox1st, 2017-07-01:
Ok Thanks :)

Pascal1573, 2017-07-02:
oke.. then I did post the wrong ones. thanks!

Ghostkeeper, 2017-07-03:
Worth noting that while Cura 2.6 has a "new" version for the UMO firmware and all its variants, there should be no change from the 2.5 other than the compilation date and the version number.

This is because of the script that I'm using to recompile: https://github.com/Ultimaker/cura-build/blob/master/marlin-build.sh
It basically just builds all of the firmwares that Cura supports except that odd Witbox one. Perhaps it should do something with the commit hashes to only recompile the ones that are updated.

Equinox1st, 2017-07-03:
Ok, so now here are the logs. 

I'm using Win 10 64 Bit and an Ultimaker 2+

First I deleted Cura 2.6.1 (the files in AppData\local an Roaming too) and installed 2.5 and did the firmware upgrade, everything was fine. 

@Pascal1573 The Logs (cura/stderr/stdout) are in the directory appdata\local 

[image]
[image]
[Log Cura 25.zip](https://github.com/Ultimaker/Cura/files/1118874/Log.Cura.25.zip)

Now i deleted Cura 2.5 (the appdata folders too) an installed 2.6.1
It abortet with the known Error Msg but whe i checked the Version on the Printer it is a new Version
@Pascal1573 The logs are here: cura.log at Roaming and the other two at local

[image]
[image]
[Log Cura 2.6.1.zip](https://github.com/Ultimaker/Cura/files/1118888/Log.Cura.2.6.1.zip)

Ghostkeeper, 2017-07-03:
I just tested this on an Ultimaker Original and an Ultimaker Original+. They seemed to upgrade fine. No error messages in Cura, just "completed".

Pascal1573, 2017-07-03:
Okay, but I still get the message. As a did mention same PC same cable same printer..
With 2.5 no error , with 2.6(.1) the commnunication error.
The firmware is loaded in the printer though!
Did you test with a apple? (Someone did mention that they did get the message on an apple)
I use the windows version.

Ghostkeeper, 2017-07-03:
It's a hackintosh: Apple hardware but the OS is Ubuntu.

fieldOfView, 2017-07-03:
(that's not a hackintosh; hackintosh is OSX on non-apple hardware)

Ghostkeeper, 2017-07-03:
Ah. Well it's a Macinhack then?

Ghostkeeper, 2017-07-07:
@teriperi tried this on Windows 10 at my behest and says it updated fine for her as well.

Pascal1573, 2017-07-09:
@Ghostkeeper:  I just updated to cura 2.6.2 and tried it again.
I still get the "error message" "Firmware update failed due to an comminication error.".
Unlike the message sugested the firmware is loaded though... firmware date 4 jul 2017.

It looks like you don't believe the other persons in this topic, only because your update is going okay? Peronaly I think you are just testing with your ubuntu box, or your running windows 10 (pro or are you using home) in a virtual machine.
(And if I'm mistaken than sorry for my comment.)

I'm going to update my windows 10 pro from annivesary edition to creators update. So I have to test eveything again, if my PC is up and running again..

Ghostkeeper, 2017-07-10:
I'm believing you! It's just that when we can't reproduce the issue on our end it's going to be very hard to do anything about it.

I was indeed testing with my Ubuntu box, but the Windows 10 was an actual Windows 10 laptop. I believe it's Enterprise edition (not entirely sure).

My prime suspect right now is the Arduino firmware. That has given us trouble in the past as well. You could try re-installing the Arduino firmware, maybe. You can install it through the Cura installer, or from the Arduino website. Don't use the default firmware that comes with Windows 10.

Pascal1573, 2017-07-19:
Sorry for the late responce.
Isn't the arduino formware/driver installed by the cura installer?
I have cura 2.5 also installed on my windows box.
Can I use or run someting from that install to install the arduino stuff?
Firmware update is going okay with the 2.5.

Ghostkeeper, 2017-07-24:
Yes the Arduino firmware should get installed by the Cura installer (unless you disable that). However if you ever tried fixing the firmware via Windows, Windows will put its own, older version of the firmware back.

The Arduino versions we delivered with 2.5 and with 2.6 are the same. It shouldn't matter. In fact, they haven't been updated in [2 years](https://github.com/Ultimaker/cura-binary-data/tree/master/windows/arduino), but still the version in Windows is older...

So if 2.5 worked and 2.6 didn't then it would be something on Cura's side that broke it. But only broke it for you.

fieldOfView, 2017-07-24:
@Ghostkeeper Firmware or driver?

Equinox1st, 2017-07-24:
@Ghostkeeper For me too 😄
can we somehow give you more Informations, so you can reproduce this? 
When 2.6.1 got out, i also tried upgrading with my win10 Laptop, this failed the same way. Sorry i forgot about it...

Equinox1st, 2017-07-25:
Ok i tried it with Cura 2.6.2. 
Laptop success - Desktop failed.... I added the Logs
[Laptop.zip](https://github.com/Ultimaker/Cura/files/1174194/Laptop.zip)
[Desktop.zip](https://github.com/Ultimaker/Cura/files/1174193/Desktop.zip)

Pascal1573, 2017-07-25:
okay....I didn't cancel the driver installation that came with the cura installer.. 
But like I mentiond my windows had an update, so I will install everything cura related once again.
(But everything I wrote was before this update (v1703 "creators update"))

The arduino driver must be the "not MS" one, so this could also mean that the usb (chipset) driver could be giving the same problem.(?) (at the moment this is a MS driver (for intel 5 series) on my machine i noticed).
Going to correct that one.

Pascal1573, 2017-07-30:
Hi. I installed the arduino driver using the newst ide (1.8.3) installer from www.arduino.cc
Sorry that I have to say that the same thing happens while updating the software.
It results in a communication error message, but it seems the firmware is loaded to the UMO  plus board.
And like before printing via USB is okay.

delinend, 2017-08-02:
I see same problem, with "Firmware update failed due to an comminication error." on my Win10 1703. 
My only solution was to use Cura 2.5.0, and then select "Upload custom Firmware", and then select the new firmware "MarlinUltimaker2plus.hex" from the "..Cura 2.6/resources/firmware" directory. This updates with no errors.
It's not the best solution, but it works until now.

Pascal1573, 2017-08-02:
Yeh that was suggested by Nallath in the past. That worked for me too , for my UMO+.
Thanks for confirming ;)

Pascal1573, 2017-08-07:
I just downloaded and installed Cura 2.7 beta from the ultimaker site.
Upgrading firmware does work in this version (2.7 beta) just like with cura 2.5 for my ultimaker Original plus.

ETitus, 2017-08-12:
Last time I printed I was using legacy Cura, time to upgrade.

Can confirm Windows 10 1703 with Cura 2.6.2 doesn't like the firmware update and it is indeed fixed in 2.7 beta, I used 2.7 to load both the firmware included in 2.7 and 2.6.2(downgraded successfully).

Ghostkeeper, 2017-08-14:
I don't remember anything being changed for 2.7 related to this.

I mean, I'm glad that it apparently works again, but I'm a bit suspicious!

TAGood827, 2017-09-05:
I will also report a failure to update using Cura 2.6.2 with Win7Pro 64bit. 

Downgrade to Cura 2.5 and it works.

Have not tried Cura 2.7 yet, will attempt after this run of parts are finished.

Equinox1st, 2017-09-05:
2.7 works fine

TAGood827, 2017-09-11:
Just tried update with Cura 2.7 and it did work.

closed 2017-09-12 by nallath