Ultimaker/Cura #2032: Marlin M205 command not being sent properly

Jailcee, 2017-06-26:
Using Cura 2.5 and 2.6, my printer is set up as a custom printer using GCode flavor "RepRap (Marlin/Sprinter)".

In Cura I have these as my settings:

Print Accel: 800

Travel Accel: 850

Print Jerk: 10

Travel Jerk: 15

According to documentation (http://marlinfw.org/docs/gcode/M204.html and http://marlinfw.org/docs/gcode/M205.html) this should produce the following Gcode:

M204 P800 T850

M205 X10 Y10

What I find in the GCode instead is:

M204 S850

M205 X15

M204 S800

M205 X10

Cura is interpreting the M205 command as if M205 Xnn  will set both X and Y jerk to nn but in fact it only sets X.  I have manually edited a Gcode file to add a M205 X10 Y10 command and saw a difference in the Y jerk (through ghosting in the print).  This confirms that the way Cura handles Jerk settings for Marlin is incorrect.

smartavionics, 2017-06-26:
Hi, have you seen this...
http://reprap.org/wiki/G-code#M205:_Advanced_settings
Perhaps marlin has changed?
The whole acceleration/jerk thing looks to be in need of some TLC as the output doesn't work for some other firmwares either.

smartavionics, 2017-06-26:
Yes, in the more recent marlin source they expect the X and Y jerk to be specified separately as you show above but what cura does is compatible with earlier marlin versions. Looking at the earlier marlin code, it would ignore a Y value if it was specified so I don't see why we can't change cura to output both X and Y values.

Jailcee, 2017-06-26:
Exactly.  The alternative is to set the max jerk settings in the eeprom but that's a bit of a hack.  Better for the Gcode to generate properly.

smartavionics, 2017-06-26:
Yes, but a potential problem is that adding the Y param to M205 could upset other printers (repetier, smoothie, etc.)

smartavionics, 2017-06-26:
OK, I think the only other firmware that could be affected is sprinter and that looks like the early marlin code so I shall put together a little PR to add the Y value...

smartavionics, 2017-06-26:
See https://github.com/Ultimaker/CuraEngine/pull/541

smartavionics, 2017-06-27:
The PR has been incorporated into the master branch so it should be in release 2.7. You can close this issue now.

closed 2017-06-27 by nallath