Ultimaker/Cura #2033: Build Plate Temperature

Bazeone, 2017-06-26:
The build plate temperature is not loaded into settings from what is setup in the material settings.

nallath, 2017-06-27:
Are you using multiple extruders with different materials? If so, it might be using a "resolve" value. As there can be only one build plate temperature, we in some cases need to do some magic (eg, if one material says 70 and the other 80, we need to do something).

Bazeone, 2017-06-27:
Only using one extruder.

Ghostkeeper, 2017-07-03:
Works for me. I added a Custom FDM Printer. The build plate temperature is 60 by default (PLA). Then I switched to CPE material and the build plate temperature becomes 70.

Bazeone, 2017-07-06:
just upgraded to 2.6.2 and the issue is still there. I'm on MacOS.

nallath, 2017-07-07:
Could you share a project file? That should contain your current setup. That way we might be able to see what's going on.

Bazeone, 2017-07-07:
The profile has the build plate set at 60, the Material is the generic ABS that is set at 80.
[AA_Arm_Model.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/1130327/AA_Arm_Model.curaproject.3mf.zip)

fieldOfView, 2017-07-07:
The profile always wins from the material.

nallath, 2017-07-07:
It also seems to have 8 extruders for some reason. So the resolve tries to take the values of those into account as wel. 
I also noticed that the custom machine plugin was acting up when I was trying to reproduce your machine; I couldn't get it to set the diameter to 1.75.

fieldOfView, 2017-07-07:
>  I couldn't get it to set the diameter to 1.75.

I'll work on that for Cura 2.7. Changing the diameter in the Machine Settings does not work because it is overridden by the diameter in the material. 

When I implemented the field in Machine Settings, the diameter specified in the XML file was still ignored. I did not recheck the Machine Settings after fixing the xml-defined diameter. 

I will fix this by checking if the material is still applicable after changing the diameter field in Machine Settings.

fieldOfView, 2017-07-07:
> It also seems to have 8 extruders for some reason. 

That is not "for some reason", that is how Custom FDM Printer is set up for multi-extrusion. It has 8 extruders defined, but uses only the first [machine_extruder_count] of them. The 3mfwriter could save some work and not export the unused extruders.

> So the resolve tries to take the values of those into account as well.

No, ExtruderManager.getExtruderValues only gets the values for extruders < machine_extruder_count

Bazeone, 2017-07-07:
what do I set the profile to for the Material temperature to win over the profile? Do I hide the setting?

Ghostkeeper, 2017-07-09:
Hiding the setting has no effect.

We don't have a way to remove a setting from a quality profile in the interface. There's basically two ways in which you could solve this:
* Switch back to Fine/Normal quality, then create a new profile with all of your original settings except the bed temperature. Unfortunately this is a lot of work.
* Go into Help -> Show Configuration Folder. Close Cura. In the configuration folder, browse to the `quality` subfolder and find the `custom_anet_a8_-_medium_quality.inst.cfg` file. Edit the file with a text editor such as notepad, and remove the line `material_bed_temperature = 60` from that file. Then start Cura again. Unfortunately, this is quite technical.

Bazeone, 2017-07-10:
That worked!, I also did it for the material diameter and that also works now too. The settings are loaded from the settings set in the materials. Would it be possible to add that I icon to set the setting back to default like some of the other settings have?

Thank you

Ghostkeeper, 2017-07-10:
While the *i* icon has a theoretical use, I think that in practice it's one of the least useful features of Cura. Perhaps it's an option to repurpose that thing to a "remove this from the quality-changes profile" or something.

fieldOfView, 2017-07-10:
I would put that functionality in the context-menu instead of further cluttering that area.

BagelOrb, 2017-07-14:
Can this issue be closed?

Bazeone, 2017-07-14:
I feel it's resolved, not user friendly, but resolved.

closed 2017-07-16 by Bazeone