Ultimaker/Cura #2036: CURA 2.6 stops (printin) briefly when reslicing or reloading

hoegge, 2017-06-27:
When you have a print running over USB and you reload a model or change parameters enough then the reslicing stops the printer printing for about 1-2 tenths of a second. Seems like this part of CURA interferes with the printer communication (using Prusa i3 MK2s on Mac)

nallath, 2017-06-27:
This is due to the GIL (global interpeter lock). Python can only do one thing at a time. We try to limit this as much as possible, but this can indeed have effect on USB printing. These kind of issues have prompted Ultimaker to drop USB serial support a long time ago.

fieldOfView, 2017-06-27:
Are you sure it is not because CuraEngine is now using all cores? Because Cura is not doing much while CuraEngine is slicing.

nallath, 2017-06-27:
It could also have an impact, but if for a very short moment the entire core that python runs on is maxed out, the USB printing will stutter.

hoegge, 2017-06-27:
For instance if you just click the infill pattern dropdown the printer freezes until you have selected or exited the dropdown (even when not changing / reslicing) anything. So this is not due to processing but some kind of deadlock in the code.

hoegge, 2017-06-27:
Whn you click top/bottom layer selector whilst printing everything stops and CPU usage by CURA goes to 0% - so it must be locking the CURA thread(s)

fieldOfView, 2017-06-28:
I think the only way to fix this then is to create USB printing host as a separate process/application (which could communicate with Cura over libArcus). I'm guessing Ultimaker is not going to implement that.

nallath, 2017-06-28:
I think it's also possible to do it with C++ & sip bindings.

fieldOfView, 2017-06-28:
Another, simpler, approach could be to use multiprocessing for usb printing to get around the GIL:
https://docs.python.org/3/library/multiprocessing.html

Ghostkeeper, 2017-07-03:
Previously cited reasons not to use Multiprocessing is because it supposedly locks up your entire computer if you're using Norton Antivirus.

ianpaschal, 2018-05-23:
As part of our new ticket handling strategy, we voted as a team on the likelihood of fixing this issue. Unfortunately, we’ve decided that we will not take time to fix it, and are marking it as `Status: Won’t Fix/Do` and closing the issue.

closed 2018-05-23 by ianpaschal

Ghostkeeper, 2018-06-29:
Norton seems to have fixed the issue. We're now using multiprocessing.

It still locks up sometimes because data needs to be transferred to the other process.