Ultimaker/Cura #2038: "Save to Removable Drive"

darkdvd, 2017-06-27:
CURA 2.6.0 / 2.6.1
Ulrimaker 2
No problem to save gcode to the SD CARD (R: Under my Windows 10 installation)
No problem to eject the SD CARD but CURA don't switch from "Save to Removable Drive" to "Save to File" as usual.
If the SD CARD is insert, CURA does not save on the SD card : must use "Save to File" to save the file on the SD CARD
No problem on previous versions since 15.x, 2.5, just with CURA 2.6.0/2.6.1

nallath, 2017-06-28:
This is because for some reason the SD card is not detected as being a removable device for some reason. Does it also not work for thumbdrives?

darkdvd, 2017-06-28:
Same problem with thumbdrive
[image]
After the SD Card OR Thumbdrives is ejected from the system, CURA continue to propose it. see screen capture.
After the SD or Thumbdrives was ejected, if I insert a second time the SD or the Thumbdrives , I am unable to click to "Save to Removable Drive" : CURA does nothing

jrdiesel76, 2017-07-02:
When I first open 2.6.1 I can save my gcode to an SD card.  If I leave the program open and want to make a change, I am unable to "save to removable drive", I have to "save to file".

fieldOfView, 2017-07-03:
Please mention your OS.
The sd issue also seems to affect (some) Windows 10 users: https://ultimaker.com/en/community/50163-cura-261-issues-sd-card-and-build-plate-temperature

Ghostkeeper, 2017-07-03:
See also this issue: https://github.com/Ultimaker/Cura/issues/2024

darkdvd, 2017-07-07:
bug est en Cura 2.6.2 also...

1nt, 2017-07-17:
In Linux Mint 18.2 Cura 2.6.2 see right SD card only on start .
If I change card, Cura does not see new card and offer to save file on the previous sd card.

rwb27, 2017-07-19:
Just in case this is useful, I have noticed since upgrading to Windows 10 and Cura 2.6.2 (both at the same time), that after I eject the removable drive it doesn't disappear from Cura, and if I put a new one in, it doesn't seem to be detected.  

Also, (and I know there's an issue #1602 for this) I used to get an error message "There is no disk in the drive. Please insert a disk into drive D:." if I started Cura with my USB card reader inserted, but without an SD card in it.  I could click "continue" a couple of times and it would go away.  Now, this message keeps reappearing until I insert a card, i.e. Cura won't start with an empty USB card reader connected.  It may be that this last change in behaviour is a Windows 10 thing, but I thought I'd mention it in case it was relevant.  After reading #1602 I suspect it's my USB dongle causing the issue, but I'll post this here anyway in case it helps someone.

I have checked with a USB stick, and the same detection issue applies - i.e. the "save to removable drive" option in Cura doesn't change to "save to file" after it's ejected.  I think that probably rules out the card reader as the sole cause of the issue for me...

nallath, 2017-07-20:
#1602 has already been fixed. That was a windows only issue.

rwb27, 2017-07-20:
Great, thanks for pointing that out.  I think I still have this issue though.  I'll update my other machine (on Win 10 already, and with the save to removable drive thing working perfectly) to 2.6.2 and see if I can reproduce the issue.

nallath, 2017-07-20:
Oh sorry, i wasn't quite clear; It's fixed for 2.7

darkdvd, 2017-07-20:
So "Save to Removable Drive" will be fixed in CURA 2.7 ? Right ?

nallath, 2017-07-20:
At least the message spam on windows is fixed. The other issue is not yet fixed, but we are looking into it. I hope to get it fixed as it's an issue that effects a lot of people.

MFHFozzy, 2017-08-31:
I just had this issue on a laptop with 2.7, running windows 10Prox64.
The laptop has an SD card slot, but no card in the slot.
Disabling the card slot in device manager allowed me to get past it.
But the issue is still there...

nallath, 2017-08-31:
Yeah, the same issue popped up but with a different reason this time. I'm sorry about all this. Now it only happens if the your SD card is in the D slot (My fix ensured that all other drives are now "safe")

ChrisTerBeke, 2018-09-28:
Old version of Cura, old ticket. Closing until problem resurfaces.

closed 2018-09-28 by ChrisTerBeke