Ultimaker/Cura #2039: 2.6.1 Values from material profile ignored

prof7bit, 2017-06-27:
The settings (temperatures) defined in the material profiles are ignored. 

When I select a different material from the materials dropdown in the upper right I expect the temperature settings (and some other meterial related settings) to be changed to the values defined in the material profile. But the temperatures do not change, The only way to manage materrials with different settings is to make different main settings profiles for different materials (introducing lots of redundancy) and ignore the materials dropdown altogether (as it has zero effect on anything).

This behavior has already been observed with 2.6.0 and it is still in 2.6.1

Ghostkeeper, 2017-07-03:
What machine and quality profile are you using?

It could very well be that the active machine is overwriting the printing temperature.

prof7bit, 2017-07-07:
I have based it on the prusa i3 preset, only changed bed dimensions. I don't see any setting for temperatures in the printer profile.

If I change the material and slice it will calculate the weight and cost correctly based on the material settings but the temperature settings in the material do not have any effect.

cmatthes, 2017-07-08:
I experienced the same issue with an Ultimaker 3 profile for the 0.8mm print cores. 
Running Cura 2.6.1 on a mac (el capitan).

Ghostkeeper, 2017-07-10:
It's not reproducing for me. I'm using the following steps:
* Add a Prusa i3 Mk2 printer.
* Go to the material manager.
* Duplicate the PLA material.
* Rename the new PLA material to BLA (just to make it recognisable).
* Change the Default Printing Temperature to 190 (down from 200).
* Activate the BLA material.

The Printing Temperature setting now says 190.0 for me.

cmatthes, 2017-07-10:
Changing the the printing temperature in the material manager works as long as you don't change it in the Print Setup menu. Clicking on the back arrow reactivates the linkage but saving the profile leads to ignored material profiles. 
In my opinion the material settings should either always update between the Print Setup menu and the material manager or you should only be able to change the setting in material manager and not the Print Setup window.

Ghostkeeper, 2017-07-11:
If you change the printing temperature in the "Print Setup" (i.e. the sidebar) then your chosen printing temperature there will always override the temperature set by the material.

The material profile is meant to be somewhat global, for re-use among many configurations, many prints, even different printers. The settings you're putting in the sidebar are meant to be specific to one set-up and perhaps just for one print, so if this one print requires a specialised temperature then you're welcome to do so but it won't also use the temperature set by the material.

So no, they aren't always in sync. The settings you set in Custom are leading. If you swap out the material, it won't change.

cmatthes, 2017-07-21:
Thanks @Ghostkeeper for the detailed explanation regarding material profiles and print settings! 

IMHO opinion the way cura handles material profiles and print settings isn't optimal at the moment.
The relationship between all the sidebar settings and material/ machines profiles isn't clear at first glance. For example there is no indication if the material settings are used or not and no way of re establishing the relationship.

Ghostkeeper, 2017-08-03:
It's mostly because the setting system is extremely complex and hard to get across. If we tried to get across then that would only really be useful information for the highest-tier power users, so we wouldn't want to clutter the interface with that sort of information.

Perhaps it could be added to the setting tooltip. Just a little line saying that currently this setting comes from the subprofile "ultimaker3-AA 0.4" in the profile "Generic PLA" or something. Think that would be an idea?

prof7bit, 2017-08-08:
The UI already has these little "i" icons that set edited but normally calculated values back to the calculated value. This was the very first thing I **intuitively** looked for but it wasn't there! Just add this "reset to calculated values" icon and add a short explanation in the tooltip, that would be all that is needeed, this would cause no UI clutter and the user experience would stay totally consistent.

Ghostkeeper, 2017-08-23:
You don't see the "i" icon if it's still a calculated value, but using a different calculation. Instead of `=default_material_print_temperature` it could now be set to `=default_material_print_temperature + 5`.

Ghostkeeper, 2017-08-23:
You don't see the "i" icon if it's still a calculated value, but using a different calculation. Instead of `=default_material_print_temperature` it could now be set to `=default_material_print_temperature + 5`.

If you overwrite that value again then you should get an "i" icon. If you click it it would reset to the value in the definition, so `=default_material_print_temperature`. This is a reset to the calculated value in the definition, but there are now multiple calculated values so it might as well restore to the calculated value from the quality profile. I think the quality profile would be more intuitive. Just remove the value from the user and the quality_changes profiles...

ianpaschal, 2018-05-22:
This has been explained and never bumped, so I think it's safe to close it.

closed 2018-05-22 by ianpaschal