Ultimaker/Cura #2040: CURA 2.6 high cpu usage

hoegge, 2017-06-27:
Is it really normal that CURA use around 25-50% CPU while printing? Assuming it just sends G-code to the printer (Prusa USB connected) - how can that take so much CPU?

nallath, 2017-06-28:
Depends on the machine. The USB code isn't al that efficient either (as it's not something we're spending any time on). It could probably be done a whole lot better, but hardly any UM machines still work on USB.

Ghostkeeper, 2017-07-03:
I can see it happening if it redraws the entire screen for every line of g-code in order to update the progress bar...

jackha, 2018-07-26:
Deferred because of inactivity.

digitalfrost, 2022-08-04:
Stale ticket.
This ticket refers to Cura 2.6 . and complains of high CPU usage.
Can be closed?

hoegge, 2022-08-08:
Most likely (can be closed). I have left Cura for Prusa Slicer a long time ago ;-)

closed 2024-04-05 by GregValiant