Ultimaker/Cura #2041: Cura 2.6 still prepends heater setting to g-code

teknikal2, 2017-06-28:
Even if I have heater settings in my start g-code, Cura still prepends the heater values to the beginning of the g-code.  I want to home the axis and move the nozzle up and off the bed before it starts heating.

fieldOfView, 2017-06-28:
What is your start gcode?

teknikal2, 2017-06-28:
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G0 Z2.0 ; move Z up a little to avoid scraping bed
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z15.0 F9000 ;15mm clearance while heating

M190 S100
M109 S235

G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
G1 F9000
M117 Printing...

teknikal2, 2017-06-28:
I had also tested (unsuccessfully) in v2.5 using the variable names from the json file here:
https://github.com/Ultimaker/Cura/blob/2.4/resources/definitions/fdmprinter.def.json

I was just hoping that maybe this was fixed with 2.6.  Just noticed that the machine start/end g-code is not being saved when I close and re-open Cura in v2.6.

fieldOfView, 2017-06-28:
Change 
```
M190 S100
M109 S235
```
to
```
M190 S{material_bed_temperature}
M109 S{material_print_temperature}
```

CuraEngine actually looks for the strings {material_bed_temperature} and {material_print_temperature} to determine wheter it should add its own heatup lines before the start gcode

fieldOfView, 2017-06-28:
>  Just noticed that the machine start/end g-code is not being saved when I close and re-open Cura in v2.6.

First: Try 2.6.1 instead of 2.6

I have heard some reports that the start/end g-code does not (always) save if you edit the textfield and then immediately close the window. Try pressing Tab after editing, and then close the window. If that fixes it I can see if I can add a proper fix for 2.7

teknikal2, 2017-06-28:
Actually, with 2.5, I think I tested {material_print_temperature_layer_0} and {material_bed_temperature_layer_0} since these seemed to be the temperature I would need to use.  I assume that if I use {material_print_temperature} and {material_bed_temperature} it will not be the special temperature that I want for the first layer.

fieldOfView, 2017-06-28:
Correct. You can also use {material_bed_temperature_layer_0} now in Cura 2.6

teknikal2, 2017-06-28:
It seems like a cleaner way to do it would be to never prepend, but instead have these lines in all the machine start g-code by default.
M190 S{material_bed_temperature_layer_0}
M109 S{material_print_temperature_layer_0}

However that might have some issues with being compatible with users' existing machine profiles...

fieldOfView, 2017-06-28:
I do agree that for learnability it would be better is those two lines would be in the default start g-code

wgaonar, 2017-08-02:
So many thanks, I am very impressed with the easy way that I can find the information about Cura and its set up. I am a user of a Mendel90 3d Printer under Manjaro Linux, and Cura is working like a charm

Ghostkeeper, 2018-04-16:
Our project manager just removed this from our planning. We won't get around to it any time soon.

ianpaschal, 2018-04-23:
Unless this issue reproduces in 2.7 and higher, we won't get to it ever and can close it.

Ghostkeeper, 2018-05-11:
There is nothing to "reproduce" since it's not a bug but an enhancement. The enhancement is still valid.

GregValiant, 2024-11-13:
Hello all.
The criteria to insert the "Prepend" heating codes was recently reworked.  Can this be closed?

closed 2024-12-10 by github-actions[bot]