Ultimaker/Cura #2042: Wrong convex hull if Horizontal Expansion < 0

fieldOfView, 2017-06-28:
While working on https://github.com/Ultimaker/Cura/pull/2031 I noticed that setting the Horizontal Expansion to a negative value, a positive offset is added to the convex hull. 

The way Polygon.getMinkowskiHull() is implemented, it can only add to the existing polygon, not subtract from it.

nallath, 2017-06-28:
So we need something like a minkowski subtraction function. 

I think it's okay-ish to leave it as it is now. It only impacts very specific situations and slightly improves packing.

Ghostkeeper, 2017-10-02:
Our project manager just removed this from our planning for being more than 12 weeks old.

I made a small patch to make sure it at least doesn't expand any more if negative, but just leave it as-is: https://github.com/Ultimaker/Cura/commit/812873af15f8d896b58fbfce6d37906fba9fb67d