Ultimaker/Cura #2043: Cura 2.6 & 2.6.1 wont start up

Demieker, 2017-06-28:
Hi folks
Am having some issues with the cura 2.6.x versions. After the cura bootscreen nothing happens.
In the taskmanager cura.exe & curaengine.exe are starting up, but then they both disappear.

I am using windows 10 version 1703, latest drivers on my dell optiplex 3240 aoi series

I cleaned my entire pc, uninstall all cura versions, deleted the users/appdata/local/cura stuff, etcetc
still get a crashlog refurring to a d:\2.6\.....   where the d: is my cdrom drive

any advice is very welcome

thanks ....

[cura26_crashlog.zip](https://github.com/Ultimaker/Cura/files/1108558/cura26_crashlog.zip)
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/1108559/DxDiag.txt)

nallath, 2017-06-28:
The logs indicate that you can start something because the machine file is corrupted. So it seems like you didn't get the right local files. It also seems to be looking for an ID i've never seen before.

Demieker, 2017-06-28:
Probably a last resort is a reinstall of windows, but i dont really want to

I had some problems with 2.5 when it was released, managed to fix it
i use different printersoftware needed: flashprint, upstudio, z-suite, bcn3d cura beta, cura 15.04.6 cura 2.xx

laptop works just fine.

kinda starts to annoy me

nallath, 2017-06-28:
A re-install of Windows is definitely not needed. I'm saying that it's crashing on some (corrupted?) local files, so it seems not everything is correctly removed. 
You need to remove the %APPDATA%\cura\2.6\ folder If I remember correctly. If that doesnt work, delete the entire %APPDATA%\cura\ folder.

fieldOfView, 2017-06-28:
Note that for Cura 2.6 the location of the configuration folder has changed. To be 100% sure, delete both C:\
Users\[you]\AppData\Local\cura (which contains the 2.5 config) and C:\
Users\[you]\AppData\Roaming\cura (which contains the 2.6 config).

Demieker, 2017-06-28:
hmmm ... i missed the c:\Users\[you]\AppData\Roaming\cura

Will try this again tomorrow.

Unclemonkeyman, 2017-06-29:
Hey guys, I have the same problem, I'm wishing that I'd stayed with 15.04.5 now as the new version i'm trying 2.6.1 takes an age to load up and eventually crashes and this is an improvement after following this thread and deleting profiles and cleaning with CCleaner reg and regular cleanup.

Demieker, 2017-06-29:
deletint both  C:\Users\[you]\AppData\Local\cura (which contains the 2.5 config) and 
C:\Users\[you]\AppData\Roaming\cura (which contains the 2.6 config).
seems to have done the trick for me.
Uninstall all versions of cura, delete both cura maps on location, running ccleaner, reinstall cura 2.6.1
so far no crashes

Unclemonkeyman, 2017-06-29:
Yep so far so good, after deleting the roaming profile.
Lets see how it goes, many thanks.

closed 2017-07-03 by Ghostkeeper