Ultimaker/Cura #2044: Cura 2.6 won't run on Mac OS X 10.10.5

mikelittle, 2017-06-28:
I tried running the latest release and it claims to require 10.11, I'm not sure why. Is there some new feature since 2.5 that the app now requires 10.11 or greater?

If this sin't just an oversight and is instead a deliberate choice, could someone point me to the area in the code where I'd need to modify it to build my own version?

HumbertoRB, 2017-06-28:
Same thing here.

nallath, 2017-06-29:
As far as I know this hasn't been a deliberate choice. We're looking into it now. Sorry for the inconvenience. I think it's our osx build server that decided to ignore our "Don't ever update" setting again (which again strengthens my believe that apple seems to hate developers) .

Ghostkeeper, 2017-07-03:
During development of version 2.7 we had to upgrade to a newer version of Qt. We suspect that something went wrong on the build server and we accidentally also upgraded the 2.6 build.

Barry37, 2017-07-06:
Cura 2.6.2 just downloaded, won't run on 10.10.5. Icon appears in dock then disappears. Not much seems to have happened, as "unidentified developer" warning appears each time. No crash report. Slightly better than 2.6 which had a "prohibited" icon.

Barry37, 2017-07-06:
Correction: this appears - Cura” is an application downloaded from the Internet. Are you sure you want to open it? not unidentified developer.

mikelittle, 2017-07-07:
@Ghostkeeper This is now fixed for 2.6.2 and works for me on OS X 10.10.5. Thanks. Happy to close this.

@Barry37 You need to click "yes" to that question and it will work from there.

closed 2017-07-07 by nallath

Barry37, 2017-07-07:
"Open"  Had done that, but hadn't moved App to Applications folder - works from there. If keeping old version, it needs to be renamed, as 2.6.2 won't run if OS X system has renamed it Cura 2

Ghostkeeper, 2017-07-09:
> Correction: this appears - Cura” is an application downloaded from the Internet. Are you sure you want to open it? not unidentified developer.

Huh, weird. We sign our MacOS builds as far as I know. Did you get this from Ultimaker's website? I'll have that investigated.

> 2.6.2 won't run if OS X system has renamed it Cura 2

Yeah, we've seen that too. It's been like that for a while. It was caused by some strangeness in how MacOS unpacks their .app files, iirc.

awhiemstra, 2017-07-10:
The "it is downloaded from the internet" message always appears for downloaded application as far as I know. It has little to do with the actual code-signing, unsigned code simply will not run (when using default OSX settings).

In regards to application naming, there are some weird things going on there where either it works after restarting, or it works after launching it a second time. The launching process on OSX is so opaque however that we never really figured out what causes this.

Barry37, 2017-07-10:
Yes, downloaded from the Ultimaker website. 
I've run previous versions renamed by OS X as Cura 2, but not from inside the Applications folder.

OS X is also quite free with the "You're opening "xyz" for the first time..." No, I'm not.

Ghostkeeper, 2017-07-13:
I've verified that the MacOS build is still signed properly.